NVlabs/ProtoMotions #2: 'scene_lib' argument missing in play_motion.py

Minyoung1005, 2024-09-27:
Hi, thanks for releasing the code for Maskedmimic. Your work is awesome!

I am trying to reproduce the results and was replaying the .npy motion files provided in the repository, and just found an error as follows when I execute `python phys_anim/scripts/play_motion.py phys_anim/data/motions/smpl_humanoid_walk.npy isaacgym smpl`

`omegaconf.errors.InterpolationKeyError: Interpolation key 'scene_lib' not found`

After adding '+scene_lib=null' to the command string in play_motion.py, it works but I'm not sure if setting scene_lib as null is appropriate.

Would appreciate your feedback, thanks!

tesslerc, 2024-09-27:
1. Good catch. We were running `play_motion.py` with the `--extra-args=...` flag and manually providing some structs. Pushed a fix to the config structure so `play_motion` will auto-load the default settings. The user can override them, for example using `--extra-args="+scenes=samp"`.
2. See the instructions in the `README` under `data`. It can receive `.npy` files, `.yaml` that points to multiple `.npy` files, or a packaged (pickled) `MotionLib` file in `.pt` format.
3. Try reducing the number of envs. Simpy add the flag `num_envs=256 algo.config.batch_size=1024`. You want to try and find the optimal working point, most envs and batch size but without it crashing due to out-of-memory errors.
4. For the SMPL one, download `SMPL+H`. For the SMPL-X body (with fingers), I've been working with `SMPL-X G`, although I believe the `N` one should also work.

tesslerc, 2024-10-17:
Marking as closed. Feel free to re-open if this is still an issue.

closed 2024-10-17 by tesslerc