NVlabs/ProtoMotions #3: Error processing AMASS

YeeLou, 2024-09-28:
Hello, I have a problem when I run `python data/scripts/convert_amass_to_isaac.py 'path/to/AMASS' --humanoid-type=smplx`.
All errors have the prompts like this:
**Error: cannot reshape array of size 115395 into shape (785,52,3)**
In fact, I found that the second dimension should be 49, not 52 as in the code. I want to know where the problem lies. Did I use the wrong smplx model file?

YeeLou, 2024-09-28:
I have tested it and found that I should download the SMPLX version. I mistakenly used the SMPLH version on certain datasets.

tesslerc, 2024-09-28:
Yes, you're correct.
For the SMPL body we use the SMPL-H files and for SMPL-X we need the full SMPL-X ones.
I'll try to clarify this in the readme.

Thanks!

closed 2024-09-28 by tesslerc