NVlabs/ProtoMotions #5: Questions regarding motion files.

Winston-Gu, 2024-09-28:
In PHC(and omnih2o), the motion file is in the format of pkl file, which may contains the whole AMASS dataset. But in this codebase, we can only use single `npy` motion file? If I want to do AMP for H1, I can only use yaml file right?

tesslerc, 2024-09-29:
Not exactly.
Our data pipeline is a bit different from PHC and derived works, but it also supports pre-processing the YAML files for faster loading.

It's explained in the data section in the readme. We have a data packaging script that creates a MotionLib object and loads the YAML file. Once loaded, it pickles the MotionLib object and saves to disk. We save this as `.pth`.

This method allows you to quickly re-load the motions and ensures you have all the data from the YAML file (for example, textual descriptions for the motions).

Winston-Gu, 2024-09-29:
I understand now. Thank you!

closed 2024-09-29 by Winston-Gu

IlanElfen, 2024-11-14:
Sorry to comment on a closed issue, but I also generated motion files from PHC. Do I have to convert my .pkl files into .yaml files. If so, how? What was the processes you used to integrate PHC/omnih2o files into ProtoMotions. Thanks!