NVlabs/ProtoMotions #6: Will you release the trained checkpoints of MaskedMimic?

hdadong, 2024-09-29:


tesslerc, 2024-09-29:
We hope to release the model weights for both the full-body tracker (phase 1) and MaskedMimic (phase 2) and are actively working towards this goal.
Hopefully we will have positive news in the next few weeks.

closed 2024-09-29 by tesslerc

hdadong, 2024-10-23:
Any new developments on this?

tesslerc, 2024-10-23:
Yes, we've released the weights earlier this week: https://github.com/NVlabs/ProtoMotions/tree/main/data/pretrained_models/MaskedMimic

See the main readme for an example how to run.

hdadong, 2024-10-24:
When i try to run the following code:
```bash
python phys_anim/eval_agent.py +robot=smplx +backbone=isaacgym +motion_file=/home/bigeast/ProtoMotions/data/amass/ACCAD/Male2MartialArtsPunches_c3d-smplx/E3____cross_left_stageii.npy  +checkpoint=/home/bigeast/ProtoMotions/data/pretrained_models/MaskedMimic/last.ckpt
```
I got the bug here:
```bash
(promotions) (base) bigeast@bigeast:~/ProtoMotions$ python phys_anim/eval_agent.py +robot=smplx +backbone=isaacgym +motion_file=/home/bigeast/ProtoMotions/data/amass/ACCAD/Male2MartialArtsPunches_c3d-smplx/E3____cross_left_stageii.npy  +checkpoint=/home/bigeast/ProtoMotions/data/pretrained_models/MaskedMimic/la
st.ckpt
Importing module 'gym_38' (/home/bigeast/Downloads/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/bigeast/Downloads/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
2024-10-24 15:27:05,913 - INFO - logger - logger initialized
phys_anim/eval_agent.py:66: UserWarning: 
The version_base parameter is not specified.
Please specify a compatability version level, or None.
Will assume defaults for version 1.1
  @hydra.main(config_path="config")
/home/bigeast/anaconda3/envs/promotions/lib/python3.8/site-packages/hydra/_internal/hydra.py:119: UserWarning: Future Hydra versions will no longer change working directory at job runtime by default.
See https://hydra.cc/docs/1.2/upgrades/1.1_to_1.2/changes_to_job_working_dir/ for more information.
  ret = run_job(
Loading training config file from /home/bigeast/ProtoMotions/data/pretrained_models/MaskedMimic/config.yaml
You are using a CUDA device ('NVIDIA GeForce RTX 4090') that has Tensor Cores. To properly utilize them, you should set `torch.set_float32_matmul_precision('medium' | 'high')` which will trade-off precision for performance. For more details, read https://pytorch.org/docs/stable/generated/torch.set_float32_matmul_precision.html#torch.set_float32_matmul_precision
Initializing distributed: GLOBAL_RANK: 0, MEMBER: 1/1
----------------------------------------------------------------------------------------------------
distributed_backend=nccl
All distributed processes registered. Starting with 1 processes
----------------------------------------------------------------------------------------------------

PyTorch version 2.2.0+cu121
Device count 1
/home/bigeast/Downloads/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/bigeast/.cache/torch_extensions/py38_cu121 as PyTorch extensions root...
Emitting ninja build file /home/bigeast/.cache/torch_extensions/py38_cu121/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
Will spawn a total of 21 objects.
Will spawn 21 scenes.
/home/bigeast/anaconda3/envs/promotions/lib/python3.8/site-packages/torch/functional.py:507: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at ../aten/src/ATen/native/TensorShape.cpp:3549.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: enabled
Creating ground plane
Ground plane created
Loading object assets... ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:--Using VHACD cache directory '/home/bigeast/.isaacgym/vhacd'
Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/Sofas/341ef2d97f0f9091183903d8843a24ef.obj'
Loading object assets... ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:--Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/Tables/b08dc9e0fd675e6664adc666ed4e2ec1.obj'
Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/LargeSofas/8918d572cff6b15df36ecf951968a8b0.obj'
Loading object assets... ━━━╸━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━  10% 0:00:01Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/LargeSofas/e7d0920ba8d4b1be71424c004dd7ab2f.obj'
Loading object assets... ━━━━━╸━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━  14% 0:00:02Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/StraightChairs/52d41eb69bb764e784aeb682b234df80.obj'
Loading object assets... ━━━━━━━╸━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━  19% 0:00:02Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/LargeSofas/277231dcb7261ae4a9fe1734a6086750.obj'
Loading object assets... ━━━━━━━━━╸━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━  24% 0:00:05Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/Armchairs/49a60133d6b05b8d19fb4103277a6b93.obj'
Loading object assets... ━━━━━━━━━━━╺━━━━━━━━━━━━━━━━━━━━━━━━━━━━  29% 0:00:08Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/Sofas/8918d572cff6b15df36ecf951968a8b0.obj'
Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/LowStools/b08dc9e0fd675e6664adc666ed4e2ec1.obj'
Loading object assets... ━━━━━━━━━━━━━━━╺━━━━━━━━━━━━━━━━━━━━━━━━  38% 0:00:07Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/Sofas/277231dcb7261ae4a9fe1734a6086750.obj'
[... cut ...]
Error executing job with overrides: ['+robot=smplx', '+backbone=isaacgym', '+motion_file=/home/bigeast/ProtoMotions/data/amass/ACCAD/Male2MartialArtsPunches_c3d-smplx/E3____cross_left_stageii.npy', '+checkpoint=/home/bigeast/ProtoMotions/data/pretrained_models/MaskedMimic/last.ckpt']
Error in call to target 'phys_anim.envs.masked_mimic.isaacgym.MaskedMimicHumanoid':
ValueError("'L_Hand' is not in list")
full_key: env

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
```

hdadong, 2024-10-24:
Then I delete the code related to 'L_hand' and 'R_hand' in data/pretrained_models/config.yaml, 
I got the another bug here:
```base
(promotions) (base) bigeast@bigeast:~/ProtoMotions$ python phys_anim/eval_agent.py +robot=smplx +backbone=isaacgym +motion_file=/home/bigeast/ProtoMotions/data/amass/ACCAD/Female1General_c3d-smplx/A1___Stand_stageii.npy   +checkpoint=/home/bigeast/ProtoMotions/data/pretrained_models/MaskedMimic/last.ckpt
Importing module 'gym_38' (/home/bigeast/Downloads/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/bigeast/Downloads/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
2024-10-24 15:49:14,209 - INFO - logger - logger initialized
phys_anim/eval_agent.py:66: UserWarning: 
The version_base parameter is not specified.
Please specify a compatability version level, or None.
Will assume defaults for version 1.1
  @hydra.main(config_path="config")
/home/bigeast/anaconda3/envs/promotions/lib/python3.8/site-packages/hydra/_internal/hydra.py:119: UserWarning: Future Hydra versions will no longer change working directory at job runtime by default.
See https://hydra.cc/docs/1.2/upgrades/1.1_to_1.2/changes_to_job_working_dir/ for more information.
  ret = run_job(
Loading training config file from /home/bigeast/ProtoMotions/data/pretrained_models/MaskedMimic/config.yaml
You are using a CUDA device ('NVIDIA GeForce RTX 4090') that has Tensor Cores. To properly utilize them, you should set `torch.set_float32_matmul_precision('medium' | 'high')` which will trade-off precision for performance. For more details, read https://pytorch.org/docs/stable/generated/torch.set_float32_matmul_precision.html#torch.set_float32_matmul_precision
Initializing distributed: GLOBAL_RANK: 0, MEMBER: 1/1
----------------------------------------------------------------------------------------------------
distributed_backend=nccl
All distributed processes registered. Starting with 1 processes
----------------------------------------------------------------------------------------------------

PyTorch version 2.2.0+cu121
Device count 1
/home/bigeast/Downloads/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/bigeast/.cache/torch_extensions/py38_cu121 as PyTorch extensions root...
Emitting ninja build file /home/bigeast/.cache/torch_extensions/py38_cu121/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
Will spawn a total of 21 objects.
Will spawn 21 scenes.
/home/bigeast/anaconda3/envs/promotions/lib/python3.8/site-packages/torch/functional.py:507: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at ../aten/src/ATen/native/TensorShape.cpp:3549.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: enabled
Creating ground plane
Ground plane created
Loading object assets... ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:--Using VHACD cache directory '/home/bigeast/.isaacgym/vhacd'
Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/HighStools/758649ba384b28856dc24db120ad1ab9.obj'
Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/LargeSofas/c2b65540d51fada22bfa21768064df9c.obj'
Loading object assets... ━╸━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   5% -:--:--Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/Sofas/8918d572cff6b15df36ecf951968a8b0.obj'
Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/Armchairs/49a60133d6b05b8d19fb4103277a6b93.obj'
Loading object assets... ━━━━━╸━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━  14% 0:00:01Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/LargeSofas/e7d0920ba8d4b1be71424c004dd7ab2f.obj'
Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/Sofas/341ef2d97f0f9091183903d8843a24ef.obj'
Loading object assets... ━━━━━━━━━╸━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━  24% 0:00:03Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/Tables/a4e5a85d21a6b36ba60a29091f2ab01f.obj'
Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/StraightChairs/243ab0d808fae5b76f1d1c49624bcfed.obj'
Loading object assets... ━━━━━━━━━━━━━╺━━━━━━━━━━━━━━━━━━━━━━━━━━  33% 0:00:02Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/LargeSofas/341ef2d97f0f9091183903d8843a24ef.obj'
Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/LowStools/a4e5a85d21a6b36ba60a29091f2ab01f.obj'
Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/LargeSofas/277231dcb7261ae4a9fe1734a6086750.obj'
Loading object assets... ━━━━━━━━━━━━━━━━━━━╺━━━━━━━━━━━━━━━━━━━━  48% 0:00:02Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/LowStools/614b46cfcbe04994d91158b2b7873a40.obj'
Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/test/SAMP/Armchairs/11040f463a3895019fb4103277a6b93.obj'
[... cut ...]
        size mismatch for vae_prior.trunk.extra_input_models.masked_mimic_target_poses.running_obs_norm.var: copying a param with shape torch.Size([184]) from checkpoint, the shape in current model is torch.Size([132]).
        size mismatch for vae_prior.trunk.extra_input_models.masked_mimic_target_poses.trunk.0.weight: copying a param with shape torch.Size([256, 184]) from checkpoint, the shape in current model is torch.Size([256, 132]).

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
```

It seem the network shape of the checkpoint is not align with the predefine value in config.

tesslerc, 2024-10-24:
Just checking -- is it working ok or are you encountering any issues running?
I saw there was some issue, but you've since edited.

hdadong, 2024-10-24:
I have fixed the bug by change the smplx to smpl and successfully run the code. But i still don't know how to control the humanoid by language or keyboard input.

tesslerc, 2024-10-24:
You can't control using the keyboard. What you can do in "user control" mode is specify some constraints through a YAML file.
I hope to add direct keyboard/mouse control in the future.

There's some more info on the user-control scheme in this thread: https://x.com/ChenTessler/status/1847923677020471799