NVlabs/ProtoMotions #7: Errors regarding path follower task.

Winston-Gu, 2024-09-30:
I am training the task of path following. The training seems normal, but when evaluating the policies, error occurs: 
"Error in call to target 'phys_anim.envs.path_following.isaacgym.PathFollowingHumanoid':
AttributeError("'PathFollowingHumanoid' object has no attribute '_num_traj_samples'")
full_key: env

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace."

My training command is 
python phys_anim/train_agent.py +exp=path_follower motion_file="phys_anim/data/motions/amp_humanoid_walk.npy" +backbone=isaacgym +robot=amp +opt=wdb num_envs=1024 algo.config.discriminator_batch_size=1024 algo.config.batch_size=4096

and my evaluating command is:
python phys_anim/eval_agent.py +exp=path_follower +backbone=isaacgym +motion_file="phys_anim/data/motions/amp_humanoid_walk.npy" +checkpoint=results/TEST/lightning_logs/version_0/last.ckpt

is there something wrong with the command I am using for evaluating the ckpt? Or is there some bugs in the script? Thanks so much for your help!

tesslerc, 2024-09-30:
Thanks for the quick heads up.
Pushed a fix, please let me know if you still encounter any issues.

For eval, we load the config file that sits alongside your `.ckpt` file.
It's recommended not to provide the exp again. This allows you to override defaults through the command line.

`python phys_anim/eval_agent.py +backbone=isaacgym +motion_file="phys_anim/data/motions/amp_humanoid_walk.npy" +checkpoint=results/TEST/lightning_logs/version_0/last.ckpt`

Winston-Gu, 2024-09-30:
Thanks mate! Solved my problem!

closed 2024-09-30 by Winston-Gu