NVlabs/ProtoMotions #8: Can training be parallelized

LanSaber, 2024-10-07:
I saw the default batch_size is 16384, which is too large for most GPUs. Therefore, I wonder if the training process can be done in parallel.

tesslerc, 2024-10-08:
Yes indeed!
The codebase builds on PyTorch Lightning (Fabric) -- https://lightning.ai/docs/fabric/stable/
In our experiments we use a SLURM managed cluster and have tested on multi-node and multi-gpu.
I haven't tried launching the experiment directly on the machine, but I believe it should work either using `torchrun` https://pytorch.org/docs/stable/elastic/run.html or using `fabric run`.

Set the flags `nodes=${nodes} ngpu=${gpu_per_node}` with nodes defining how many distinct compute nodes (machines) you use and gpu-per-node the number of GPUs on each machine.

Keep in mind that the flags are per-rank! So if you set `num_envs=4096` you will have 4096 environments on each GPU. Same with batch size.

tesslerc, 2024-10-17:
Marking as closed. Feel free to re-open if this is still an issue.

closed 2024-10-17 by tesslerc