NVlabs/ProtoMotions #9: Early stop during training

YeeLou, 2024-10-07:
Hello, thank you for your good work!
There was an early stop while I was training my full body tracker:
> `Epoch 1800, collecting data... ━━━━━━━━━━━━━━━━━━━━━━━━━━╺━━━━━━━━━━━━━  66% 0:00:03 PxgCudaDeviceMemoryAllocator fail to allocate memory 1610612736 bytes!! Result = 2
/buildAgent/work/99bede84aa0a52c2/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp (2155) : internal error : memcpy failed fail! 2  700 

/buildAgent/work/99bede84aa0a52c2/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp (2165) : internal error : GPU cudaMainGjkEpa or prepareLostFoundPairs kernel fail!
  700

/buildAgent/work/99bede84aa0a52c2/source/gpucommon/include/PxgCudaUtils.h (54) : internal error : SynchronizeStreams cuEventRecord failed

/buildAgent/work/99bede84aa0a52c2/source/gpucommon/include/PxgCudaUtils.h (60) : internal error : SynchronizeStreams cuStreamWaitEvent failed`

Due to my GPU limitations, I added the flag: > num_envs=256 algo.config.batch_size=1024. When I change the flag to: > num_envs=256 algo.config.batch_size=512, The error still occurs.

Could you please tell me what I should do to solve this problem?

tesslerc, 2024-10-08:
The full body tracker experiment uses a complex terrain by default.
Try setting the flag `force_flat_terrain=True`.
If that solves the issue and you do want to train with terrains, try editing the terrain parameters in `config/terrain/complex.yaml`. Less number of levels/maps should result in a smaller terrain and less memory consumption.

YeeLou, 2024-10-09:
Thank you! Additionally, I was wondering which datasets from AMASS were used specifically for training the full body tracker?

tesslerc, 2024-10-09:
We used the entire AMASS dataset and took all motions with a text description in HML3D and for those we discarded the unlabeled portions.
I believe this covers the majority of the AMASS dataset. You can see the scripts in the data folder, they should reproduce the exact same data we use for training.

tesslerc, 2024-10-17:
Marking as closed. Feel free to re-open if this is still an issue.

bowieshi, 2024-11-23:
@YeeLou ,Hi. I am also trying to train the full body tracker model. But I met a problem of training on multi-GPU. My training process is just stuck in `self.actor, self.actor_optimizer = self.fabric.setup(actor, actor_optimizer)`. Can I ask what is your full training command? Thank you.