NVlabs/ProtoMotions #10: About the Unified Controller for Mimicking AMASS Motions

thyzju17, 2024-10-09:
Great job! 
I'm particularly intrigued by Stage 1, which mimics all motions in AMASS. 
In SuperPADL, different controllers are trained for each motion clip and then distilled into a unified controller, suggesting that mimicking all motions with a single controller in DRL is challenging. However, in your work, it appears to be successful. Could it be that the transformer structure contributes to this achievement?

tesslerc, 2024-10-09:
- I think the success is due to (1) the training objectives are slightly different, (2) the architecture is bigger, (3) we dynamically prioritize harder samples during training.
- PHC+ (https://github.com/ZhengyiLuo/PHC) also reaches 99.9% or higher on the train set with a single controller.
- We do see higher success rate on the unseen data with our model.

So the transformer might be helpful in improving generalization (test set) compared to a fully-connected model, but the big leap over SuperPADL is probably attributed more to the training setup (more learnable parameters + prioritizing hard motions).

thyzju17, 2024-10-09:
Thank you so much for your time!

closed 2024-10-09 by thyzju17