NVlabs/ProtoMotions #17: Will you release the trained checkpoints of path following?

leele-bing, 2024-10-29:
Hello, thank you for the great work!
We are interested in the path following with the SMPL robot and ask if you can provide the checkpoints for open source.

tesslerc, 2024-10-29:
We don't have pretrained models we can share for this task.
If possible, I suggest training one yourself. Training should be relatively fast on desktop hardware (few days).
If you have issues setting up the data/training, please let me know and I'll try my best to help out.

leele-bing, 2024-11-04:
Thanks！
Can you tell me how to train a SMPL Robot to follow a path on flat ground based only on AMP?

tesslerc, 2024-11-05:
`python phys_anim/train_agent.py +exp=path_follower +robot=smpl +backbone=<simulator> +env.config.path_follower_params.height_conditioned=False +env.config.path_follower_params.path_generator.use_naive_path_generator=True force_flat_terrain=True motion_file=<motionfilepath>`

- Turning off `height_conditioned` will make it only walk upright. If it's enabled it trains to crouch-walk and even crawl. But it needs corresponding data.
- Enabling the `naive_path_generator` will randomly sample paths that do not depend on the requested head-height.
- Setting `force_flat_terrain` will spawn a simple flat ground with the minimal triangle mesh representation, instead of the default complex irregular terrain.

closed 2024-11-11 by tesslerc