NVlabs/ProtoMotions #19: torch install

Kirovsiki, 2024-11-07:
HI,Thank you very much for your willingness to open source your project, but I'm having conflict issues configuring my environment. The code will install torch2.4.0 first, then uninstall and install 2.2.0 which doesn't support cuda. how should I adapt to the environment? Looking forward to your reply!

arslanr369, 2024-11-07:
To resolve the issue:

1. **Check CUDA compatibility**: Ensure the PyTorch version matches your CUDA version.
2. **Use a clean environment**: Create a new `conda` or `virtualenv` environment to avoid conflicts.
3. **Directly install the right version**: Install PyTorch 2.4.0 with CUDA support, like:
   ```bash
   pip install torch==2.4.0+cuXX
   ```
4. **Avoid unnecessary uninstalls**: Modify the script to install only the required version (2.4.0 with CUDA).

This should help prevent version conflicts and ensure proper CUDA support.

Kirovsiki, 2024-11-07:
Yes, it seems that our isaac's python environment will be broken after this project operation, and then the code will not be able to find the rest of isaac.

tesslerc, 2024-11-07:
It depends on your python version.
If you're using 3.8, then the latest supported pytorch lightning version is 2.3.3 which supports torch 2.3.
If you have 3.9 or higher, then you can go with the latest pytorch lightning 2.4 and that does have support for torch 2.4.

On my setup it's with torch 2.2.2 and I haven't encountered missing CUDA support.

Kirovsiki, 2024-11-08:
Thank you very much for your reply, I'm trying to install with a older version of iscca

Kirovsiki, 2024-11-08:
I installed iscca version 4.1.0 and looked for his environment to be python3.10.14, I should probably try 2.4.0 torch.

Kirovsiki, 2024-11-08:
After executing this step
PYTHON_PATH -m pip install -e .
I get these errors: ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behavior results in the following dependency conflict.
nvidia-srl-usd 0.14.0 requires usd-core<24.00,>=21.11, which is not yet installed.
nvidia-srl-usd-to-urdf 0.6.0 requires usd-core<24.00,>=21.11, which is not yet installed.
nvidia-srl-usd-to-urdf 0.6.0 requires lxml<5.0.0,>=4.9.2, which your lxml 5.2.2 is not compatible with.
torchvision 0.17.2+cu118 requires torch==2.2.2+cu118, but your torch 2.2.0 is not compatible.
torchaudio 2.2.2+cu118 requires torch==2.2.2+cu118, but your torch 2.2.0 is not compatible.

Kirovsiki, 2024-11-08:
Error executing job with overrides: ['+exp=full_body_tracker', '+robot=smpl', '+backbone=isaacsim', 'motion_file=C:\\Users\\Administrator\\Desktop\\TestMT\\ProtoMotions-main\\phys_anim\\data\\motions\\smplx_humanoid_walk.npy']
Error in call to target 'lightning.fabric.fabric.Fabric':
RuntimeError('No supported gpu backend found!')
full_key: fabric
I seem to be experiencing a new problem, I switch off multicard and start training on the local gpu and then experience an inability to find the GPU.

Kirovsiki, 2024-11-08:
It looks like I'm missing something, what is it? I can't seem to find a description of it:Error executing job with overrides: ['+exp=full_body_tracker', '+robot=smpl', '+backbone=isaacsim', 'motion_file=C:\\Users\\Administrator\\Desktop\\TestMT\\ProtoMotions-main\\phys_anim\\data\\motions\\smplx_humanoid_walk.npy']
Traceback (most recent call last):
  File "C:\Users\Administrator\Desktop\TestMT\ProtoMotions-main\phys_anim\train_agent.py", line 116, in main
    simulation_app = SimulationApp(
  File "E:\TestON\isaac-sim-4.0.0/exts/omni.isaac.kit\omni\isaac\kit\simulation_app.py", line 250, in __init__
    from omni.kit.window.title import get_main_window_title
ModuleNotFoundError: No module named 'omni.kit.window.title'

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
2024-11-08 07:46:30 [3,303ms] [Warni

Kirovsiki, 2024-11-08:
Looking forward to your answer，I seem to be having strange dependency issues, is this version too old：Error executing job with overrides: ['+exp=full_body_tracker', '+robot=smpl', '+backbone=isaacsim', 'motion_file=C:\\Users\\Administrator\\Desktop\\TestMT\\ProtoMotions-main\\phys_anim\\data\\motions\\smplx_humanoid_walk.npy']
Traceback (most recent call last):
  File "C:\Users\Administrator\Desktop\TestMT\ProtoMotions-main\phys_anim\train_agent.py", line 116, in main
    simulation_app = SimulationApp(
  File "E:\TestON\isaac-sim-4.0.0/exts/omni.isaac.kit\omni\isaac\kit\simulation_app.py", line 250, in __init__
    from omni.kit.window.title import get_main_window_title
ModuleNotFoundError: No module named 'omni.kit.window.title'

Kirovsiki, 2024-11-11:
I seem to be having weird dependency issues, this looks like another visualization module, do I need to change to headless mode :Error executing job with overrides: ['+exp=full_body_tracker', '+robot=smpl', '+backbone=isaacsim', 'motion_file=C:\\Users\\Administrator\\Desktop\\TestMT\\ProtoMotions-main\\phys_anim\\data\\motions\\smplx_humanoid_walk.npy']
Traceback (most recent call last):
  File "C:\Users\Administrator\Desktop\TestMT\ProtoMotions-main\phys_anim\train_agent.py", line 116, in main
    simulation_app = SimulationApp(
  File "E:\TestON\isaac-sim-4.0.0/exts/omni.isaac.kit\omni\isaac\kit\simulation_app.py", line 250, in __init__
    from omni.kit.window.title import get_main_window_title
ModuleNotFoundError: No module named 'omni.kit.window.title'

tesslerc, 2024-11-11:
I'm not sure, the IsaacSim integration is relatively new so I have less experience with the nuances of it.
I am working on converting from IsaacSim to IsaacLab. It's an extensions on-top of IsaacSim that simplifies some of the workflow.
I hope to share that code in the near future.

Kirovsiki, 2024-11-13:
Thank you very much for your reply. We switched gym to allow it. But unfortunately our CUDA memory (64G) still doesn't seem to be able to do pretrain tasks. We'll try to experiment with a bigger graphics card.

closed 2024-11-20 by tesslerc