NVlabs/ProtoMotions #20: ModuleNotFoundError: No module named 'phys_anim.data.assets.skeleton_configs'

bowieshi, 2024-11-11:
Hi! Thanks for your great work!
I am trying to train the stage one full body tracking model and I tried to process AMASS motion data to isaacgym compatible format. When I was packaging the data for faster loading:
I run the command:
```
python data/scripts/package_motion_lib.py <path_to_yaml_file> <path_to_AMASS_data_dir> <output_pt_file_path> --humanoid-type=smplx
```
But received the error:
```
Traceback (most recent call last):
  File "data/scripts/package_motion_lib.py", line 37, in <module>
    from phys_anim.data.assets.skeleton_configs import isaacgym_asset_file_to_stats
ModuleNotFoundError: No module named 'phys_anim.data.assets.skeleton_configs'
```
I check the corresponding path and can not find the corresponding module in github repository. How can I get around it? Thank you for your help!

tesslerc, 2024-11-11:
Thanks for catching this.
I've just pushed a fix. It now takes the skeleton configuration from the config files, removing the need for the deprecated skeleton_configs.py file.

Please let me know if there's any further issues!

bowieshi, 2024-11-12:
@tesslerc Hi. Thank you for your response. I also have a question about how to convert SAMP dataset. Which fille should I download from the SAMP website and in which hierarchy should I arrange the files in order to use the `data/scripts/convert_samp_to_isaac.py` to convert the dataset. Thank you!

closed 2024-11-13 by bowieshi