NVlabs/ProtoMotions #25: Question about training of full body tracker

bowieshi, 2024-11-15:
Thank you for your excellent work!!
I am trying to train the full body tracker these days. I turned off the headless mode and found the humanoid's initial pose is floating above the ground. You can refer to the following isaacgym rendering result. The humanoid is about 20cm higher the tracking joints points. Can I inquire whether this is normal?
[image]
[image]

Also, I want to inquire that how long will the training process last if I only want to train the first stage full body tracking policy. Will the pretrained checkpoints of the first stage full body tracking policy be open-sourced? Thank you again!

tesslerc, 2024-11-15:
(1)
I assume you're checking the kinematic replay.
Kinematic replay is essentially respawning the humanoid every frame to the next pose from the data. By default, we respawn the humanoid with an offset of 20cm to ensure there's no mesh collisions. This is also applied during kinematic replay, as collisions might lead to weird artifacts since physics is actually active.

You can turn this off with the flag `ref_respawn_offset=0`, but keep in mind that it might still have a small offset when located above an object, due to how we solve collisions.

When training and when running inference it will need to track the locations of the red spheres. They are the ground truth motion and where the humanoid will be. So it's not really an issue its spawned with 20cm offset as it meets the ground within 2-3 frames.

(2)
I'll look into releasing the weights for the first stage, but I'm not sure what the timeframe would be.

(3)
Really exciting to see you have the full pipeline converted and working!

dbdxnuliba, 2024-11-19:
@bowieshi ，could you please tell me about the full command to train ，and by the way,what model of graphics card are you using

bowieshi, 2024-11-21:
> @bowieshi ，could you please tell me about the full command to train ，and by the way,what model of graphics card are you using

I am using the following command. I used A100 to train the model. I still met some errors about insufficient GPU memory. So I can only run 16 environments. I am still trying to locate the problem.
```
python phys_anim/train_agent.py \
    +exp=full_body_tracker \
    +robot=smplx \
    +backbone=isaacgym \
    motion_file=AMASS-test/motion_file.pt \
    force_flat_terrain=True \
    algo.config.batch_size=16 \
    num_envs=16 \
    headless=False
```

tesslerc, 2024-11-21:
That seems terribly low.
I'd expect at least 1024 envs on a 40gb GPU and up to 4096 with the 80gb version.

closed 2024-11-24 by bowieshi