NVlabs/ProtoMotions #26: Nice and well-organized codebase. One question.

YanjieZe, 2024-11-17:
Hi! Thank you for open-sourcing the code. I have one dumb question.
In the commands recorded in the README, they all use only one motion file. To train my own full whole-body tracking controller, should I download all AMASS data and compress them into one .npy file, and then train on it?
Thank you very much for your reply.

tesslerc, 2024-11-18:
To load multiple files, we define the structure using a yaml file. See [this file](https://github.com/NVlabs/ProtoMotions/blob/main/data/yaml_files/amass_train.yaml) for an example.

In the Data section in the readme there's an explanation how to pre-load the dataset and store it as a single reusable `.pt` file.

YanjieZe, 2024-11-19:
Hi, as you suggest, I am trying to generate H1 motion files with given scripts and then generate my own yaml files.   I encounter this problem when I was trying to generate my yaml file on my generated H1 motions:                
[image]
Have you tested the script to generate yaml files on H1 and encountered similar issues?

tesslerc, 2024-11-19:
Yes.
You can manually iterate over all the `.npy` files created in the retargeting process and create your own yaml file.
HML3D is optional and provides text labels + breaks the motions into sub-motions with semantic meanings. If it's not provided, the full motion file will be used as a single sequence and no text embeddings will be generated.

```
motions:
- file: KIT-h1/3/jump_left02_poses.npy
  fps: 30
  idx: 0
  weight: 1.0
- file: KIT-h1/1747/displace_from_left_to_right_06_poses.npy
  fps: 30
  idx: 1
  weight: 1.0
...
```

Basically:
1. Create a list of all motion files.
2. `idx` should be unique and increasing. this is mainly for sanity checks in the code, which may be redundant, but are required at this point.
3. `fps` for all the converted H1 motions should be 30 FPS (as handled by the conversion script).
4. `weight` will be optional in the future. for now you can just set it to `1.0` for all motions.

YanjieZe, 2024-11-19:
Thank you! This is informative.

closed 2024-11-19 by YanjieZe