NVlabs/ProtoMotions #29: error train h1 full_body_tracker

dbdxnuliba, 2024-11-19:
(ProtoMotions) rob@rob:~/rl/ProtoMotions$ alias PYTHON_PATH=python
(ProtoMotions) rob@rob:~/rl/ProtoMotions$ PYTHON_PATH phys_anim/train_agent.py +exp=full_body_tracker +robot=h1 +backbone=isaacgym motion_file=phys_anim/data/motions/h1_punch.npy num_envs=512 algo.config.batch_size=1024 force_flat_terrain=True headless=false
Importing module 'gym_38' (/home/rob/rl/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/rob/rl/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
2024-11-19 13:07:07,096 - INFO - logger - logger initialized
phys_anim/train_agent.py:71: UserWarning: 
The version_base parameter is not specified.
Please specify a compatability version level, or None.
Will assume defaults for version 1.1
  @hydra.main(config_path="config", config_name="base")
/home/rob/anaconda3/envs/ProtoMotions/lib/python3.8/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'base': Defaults list is missing `_self_`. See https://hydra.cc/docs/1.2/upgrades/1.0_to_1.1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
/home/rob/anaconda3/envs/ProtoMotions/lib/python3.8/site-packages/hydra/_internal/hydra.py:119: UserWarning: Future Hydra versions will no longer change working directory at job runtime by default.
See https://hydra.cc/docs/1.2/upgrades/1.1_to_1.2/changes_to_job_working_dir/ for more information.
  ret = run_job(
Initializing distributed: GLOBAL_RANK: 0, MEMBER: 1/1
----------------------------------------------------------------------------------------------------
distributed_backend=nccl
All distributed processes registered. Starting with 1 processes
----------------------------------------------------------------------------------------------------

[rank: 0] Seed set to 0
Setting seed: 0
PyTorch version 2.2.2+cu121
Device count 1
/home/rob/rl/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/rob/.cache/torch_extensions/py38_cu121 as PyTorch extensions root...
Emitting ninja build file /home/rob/.cache/torch_extensions/py38_cu121/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
Will spawn a total of 339 objects.
Will spawn 339 scenes.
/home/rob/anaconda3/envs/ProtoMotions/lib/python3.8/site-packages/torch/functional.py:507: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at ../aten/src/ATen/native/TensorShape.cpp:3549.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: enabled
Creating ground plane
Ground plane created
Error executing job with overrides: ['+exp=full_body_tracker', '+robot=h1', '+backbone=isaacgym', 'motion_file=phys_anim/data/motions/h1_punch.npy', 'num_envs=512', 'algo.config.batch_size=1024', 'force_flat_terrain=True', 'headless=false']
Error in call to target 'phys_anim.envs.mimic.isaacgym.MimicHumanoid':
MissingMandatoryValue('Missing mandatory value: config.robot.num_bodies\n    full_key: config.robot.num_bodies\n    object_type=dict')
full_key: env
Missing mandatory value: config.robot.num_bodies
    full_key: config.robot.num_bodies
    object_type=dict

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.

tesslerc, 2024-11-19:
Thanks. For the H1 we don't have the bfs joint ordering yet as we haven't integrated it into IsaacSim yet.
Pushed a fix to take the num_bodies from the IsaacGym ordering, which is available for all robots.

This is a temporary solution until we push a more robust fix for long term stability.

Let me know if you encounter any further issues!

dbdxnuliba, 2024-11-25:
thansk for your answer,
when run the following command, it occurs err:

```
(ProtoMotions) rob@rob:~/rl/ProtoMotions$ PYTHON_PATH phys_anim/train_agent.py +exp=full_body_tracker +robot=h1 +backbone=isaacgym motion_file=phys_anim/data/motions/h1_punch.npy  num_envs=1 algo.config.batch_size=1024 force_flat_terrain=True headless=false
Importing module 'gym_38' (/home/rob/rl/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/rob/rl/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
2024-11-25 14:10:15,146 - INFO - logger - logger initialized
phys_anim/train_agent.py:71: UserWarning: 
The version_base parameter is not specified.
Please specify a compatability version level, or None.
Will assume defaults for version 1.1
  @hydra.main(config_path="config", config_name="base")
/home/rob/anaconda3/envs/ProtoMotions/lib/python3.8/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'base': Defaults list is missing `_self_`. See https://hydra.cc/docs/1.2/upgrades/1.0_to_1.1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
/home/rob/anaconda3/envs/ProtoMotions/lib/python3.8/site-packages/hydra/_internal/hydra.py:119: UserWarning: Future Hydra versions will no longer change working directory at job runtime by default.
See https://hydra.cc/docs/1.2/upgrades/1.1_to_1.2/changes_to_job_working_dir/ for more information.
  ret = run_job(
Initializing distributed: GLOBAL_RANK: 0, MEMBER: 1/1
----------------------------------------------------------------------------------------------------
distributed_backend=nccl
All distributed processes registered. Starting with 1 processes
----------------------------------------------------------------------------------------------------

[rank: 0] Seed set to 0
Setting seed: 0
PyTorch version 2.2.0+cu121
Device count 1
/home/rob/rl/IsaacGym_Preview_4_Package/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/rob/.cache/torch_extensions/py38_cu121 as PyTorch extensions root...
Emitting ninja build file /home/rob/.cache/torch_extensions/py38_cu121/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
Will spawn a total of 339 objects.
Will spawn 339 scenes.
/home/rob/anaconda3/envs/ProtoMotions/lib/python3.8/site-packages/torch/functional.py:507: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at ../aten/src/ATen/native/TensorShape.cpp:3549.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: enabled
Creating ground plane
Ground plane created
Loading object assets... ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:--Using VHACD cache directory '/home/rob/.isaacgym/vhacd'
Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/train/SAMP/LargeSofas/a8db9db73bb5a4dbf51f77a6d7299806.obj'
Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/train/SAMP/HighStools/4386633c2d2a1a2c5fceaa39a8353bde.obj'
Loading object assets... ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:--Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/train/SAMP/LowStools/575ac593154b95b6dae7645f889ddcf0.obj'
Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/train/SAMP/Sofas/15ca4d51db95edd783f3ef99fe735c7e.obj'
Loading object assets... ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   1% 0:00:09Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/train/SAMP/LowStools/6d2c75d89b0a5f5d8cbb8bac2032149c.obj'
Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/train/SAMP/Armchairs/90c8ae65e3ec4ef119fb4103277a6b93.obj'
Loading object assets... ╸━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   1% 0:05:11Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/train/SAMP/HighStools/c4ee00c87688f3b0b161476a346db176.obj'
Loading object assets... ╸━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   2% 0:04:45Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/train/SAMP/Armchairs/c47e364fbc29803b2c0eb4a518e123ac.obj'
Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/train/SAMP/LowStools/80a86204247b30fd21ef666e979e4e80.obj'
Loading object assets... ╸━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   2% 0:03:34Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/train/SAMP/StraightChairs/25957008f839ef647abe6643657b8aec.obj'
Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/train/SAMP/LargeSofas/1878633a36518ffc19fb4103277a6b93.obj'
Loading object assets... ━╺━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   3% 0:02:49Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/train/SAMP/LowStools/dee160f20e1195a4b2cf6e4d01862bc5.obj'
Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/train/SAMP/Armchairs/9faefdf6814aaa975510d59f3ab1ed64.obj'
Found existing convex decomposition for mesh 'phys_anim/data/assets/scenes/train/SAMP/HighStools/8da6fc70195d2b4029820ac1e8a0acf2.obj'
[... cut ...]
  File "/home/rob/anaconda3/envs/ProtoMotions/lib/python3.8/site-packages/torch/utils/_contextlib.py", line 115, in decorate_context
    return func(*args, **kwargs)
  File "/home/rob/rl/ProtoMotions/phys_anim/agents/ppo.py", line 584, in generate_datasets
    self.actor_critic_dataset = DictDataset(
  File "/home/rob/rl/ProtoMotions/phys_anim/agents/utils/data_utils.py", line 97, in __init__
    assert (
AssertionError: 32 1024

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.

```

dbdxnuliba, 2024-11-26:
but 1024/(1*32) = 32 without remainder

tesslerc, 2024-11-26:
Yes, but you're splitting the data the agent collects into batches.
That would be `1 (envs) * 32 (steps) = data`. And then `data / batch_size = 32 / 1024`.

closed 2024-12-17 by dbdxnuliba