NVlabs/ProtoMotions #30: training with multiple GPU with problem that the code is stuck

bowieshi, 2024-11-22:
Hi. I can successfully train the code on a single GPU. After I modified the fabric.yaml file to the following, and train with command as following, my code is stuck and cannot continue to the data collection and training part. 
I am using 4 A100 to parralel the training process.  I run with 2048 num_env and batch_size of 8192. I also set the force_flat_terrain to be true

Also I want to inquire how many days you used to train a good full-body tracking policy on AMASS and samp. 

part of fabric.yaml
```
ngpu: 4
nodes: 1
torch_deterministic: False
amp_precision: 32-true
force_flat_terrain: False

eval_overrides:
  ngpu: 4
  nodes: 1

  fabric:
    loggers: null
```

training command
```
CUDA_VISIBLE_DEVICES=1,2,5,7 python phys_anim/train_agent.py \
    +exp=full_body_tracker \
    +robot=smplx \
    +backbone=isaacgym \
    motion_file=/media/xuzhao/Partition5/qyzheng/boao/AMASS-test/motion_file.pt \
    force_flat_terrain=True \
    algo.config.batch_size=8192 \
    num_envs=2048
```
The program is stuck here for a long time. So I need to kill it.
[image]

bowieshi, 2024-11-22:
At last, the following is outputted by the program and the program is killed
```
[rank2]:[E ProcessGroupNCCL.cpp:523] [Rank 2] Watchdog caught collective operation timeout: WorkNCCL(SeqNum=1, OpType=ALLGATHER, NumelIn=1, NumelOut=4, Timeout(ms)=1800000) ran for 1800130 milliseconds before timing out.
[rank2]:[E ProcessGroupNCCL.cpp:537] Some NCCL operations have failed or timed out. Due to the asynchronous nature of CUDA kernels, subsequent GPU operations might run on corrupted/incomplete data.
[rank2]:[E ProcessGroupNCCL.cpp:543] To avoid data inconsistency, we are taking the entire process down.
[rank2]:[E ProcessGroupNCCL.cpp:1182] [Rank 2] NCCL watchdog thread terminated with exception: [Rank 2] Watchdog caught collective operation timeout: WorkNCCL(SeqNum=1, OpType=ALLGATHER, NumelIn=1, NumelOut=4, Timeout(ms)=1800000) ran for 1800130 milliseconds before timing out.
Exception raised from checkTimeout at ../torch/csrc/distributed/c10d/ProcessGroupNCCL.cpp:525 (most recent call first):
frame #0: c10::Error::Error(c10::SourceLocation, std::string) + 0x57 (0x7f3252381d87 in /home/qyzheng/anaconda3/envs/isaacgym/lib/python3.8/site-packages/torch/lib/libc10.so)
frame #1: c10d::ProcessGroupNCCL::WorkNCCL::checkTimeout(std::optional<std::chrono::duration<long, std::ratio<1l, 1000l> > >) + 0x1e6 (0x7f315a60c6e6 in /home/qyzheng/anaconda3/envs/isaacgym/lib/python3.8/site-packages/torch/lib/libtorch_cuda.so)
frame #2: c10d::ProcessGroupNCCL::workCleanupLoop() + 0x19d (0x7f315a60fc3d in /home/qyzheng/anaconda3/envs/isaacgym/lib/python3.8/site-packages/torch/lib/libtorch_cuda.so)
frame #3: c10d::ProcessGroupNCCL::ncclCommWatchdog() + 0x119 (0x7f315a610839 in /home/qyzheng/anaconda3/envs/isaacgym/lib/python3.8/site-packages/torch/lib/libtorch_cuda.so)
frame #4: <unknown function> + 0xdbbf4 (0x7f3254ac7bf4 in /home/qyzheng/anaconda3/envs/isaacgym/bin/../lib/libstdc++.so.6)
frame #5: <unknown function> + 0x94b43 (0x7f325ad67b43 in /lib/x86_64-linux-gnu/libc.so.6)
frame #6: <unknown function> + 0x126a00 (0x7f325adf9a00 in /lib/x86_64-linux-gnu/libc.so.6)

[rank1]:[E ProcessGroupNCCL.cpp:523] [Rank 1] Watchdog caught collective operation timeout: WorkNCCL(SeqNum=1, OpType=ALLGATHER, NumelIn=1, NumelOut=4, Timeout(ms)=1800000) ran for 1800259 milliseconds before timing out.
[rank1]:[E ProcessGroupNCCL.cpp:537] Some NCCL operations have failed or timed out. Due to the asynchronous nature of CUDA kernels, subsequent GPU operations might run on corrupted/incomplete data.
[rank1]:[E ProcessGroupNCCL.cpp:543] To avoid data inconsistency, we are taking the entire process down.
[rank1]:[E ProcessGroupNCCL.cpp:1182] [Rank 1] NCCL watchdog thread terminated with exception: [Rank 1] Watchdog caught collective operation timeout: WorkNCCL(SeqNum=1, OpType=ALLGATHER, NumelIn=1, NumelOut=4, Timeout(ms)=1800000) ran for 1800259 milliseconds before timing out.
Exception raised from checkTimeout at ../torch/csrc/distributed/c10d/ProcessGroupNCCL.cpp:525 (most recent call first):
frame #0: c10::Error::Error(c10::SourceLocation, std::string) + 0x57 (0x7f37e36f4d87 in /home/qyzheng/anaconda3/envs/isaacgym/lib/python3.8/site-packages/torch/lib/libc10.so)
frame #1: c10d::ProcessGroupNCCL::WorkNCCL::checkTimeout(std::optional<std::chrono::duration<long, std::ratio<1l, 1000l> > >) + 0x1e6 (0x7f36eb80c6e6 in /home/qyzheng/anaconda3/envs/isaacgym/lib/python3.8/site-packages/torch/lib/libtorch_cuda.so)
frame #2: c10d::ProcessGroupNCCL::workCleanupLoop() + 0x19d (0x7f36eb80fc3d in /home/qyzheng/anaconda3/envs/isaacgym/lib/python3.8/site-packages/torch/lib/libtorch_cuda.so)
frame #3: c10d::ProcessGroupNCCL::ncclCommWatchdog() + 0x119 (0x7f36eb810839 in /home/qyzheng/anaconda3/envs/isaacgym/lib/python3.8/site-packages/torch/lib/libtorch_cuda.so)
frame #4: <unknown function> + 0xdbbf4 (0x7f37e5cc7bf4 in /home/qyzheng/anaconda3/envs/isaacgym/bin/../lib/libstdc++.so.6)
frame #5: <unknown function> + 0x94b43 (0x7f37ebda2b43 in /lib/x86_64-linux-gnu/libc.so.6)
frame #6: <unknown function> + 0x126a00 (0x7f37ebe34a00 in /lib/x86_64-linux-gnu/libc.so.6)

[rank3]:[E ProcessGroupNCCL.cpp:523] [Rank 3] Watchdog caught collective operation timeout: WorkNCCL(SeqNum=1, OpType=ALLGATHER, NumelIn=1, NumelOut=4, Timeout(ms)=1800000) ran for 1800290 milliseconds before timing out.
[rank3]:[E ProcessGroupNCCL.cpp:537] Some NCCL operations have failed or timed out. Due to the asynchronous nature of CUDA kernels, subsequent GPU operations might run on corrupted/incomplete data.
[rank3]:[E ProcessGroupNCCL.cpp:543] To avoid data inconsistency, we are taking the entire process down.
[rank3]:[E ProcessGroupNCCL.cpp:1182] [Rank 3] NCCL watchdog thread terminated with exception: [Rank 3] Watchdog caught collective operation timeout: WorkNCCL(SeqNum=1, OpType=ALLGATHER, NumelIn=1, NumelOut=4, Timeout(ms)=1800000) ran for 1800290 milliseconds before timing out.
Exception raised from checkTimeout at ../torch/csrc/distributed/c10d/ProcessGroupNCCL.cpp:525 (most recent call first):
frame #0: c10::Error::Error(c10::SourceLocation, std::string) + 0x57 (0x7fe4d4b81d87 in /home/qyzheng/anaconda3/envs/isaacgym/lib/python3.8/site-packages/torch/lib/libc10.so)
frame #1: c10d::ProcessGroupNCCL::WorkNCCL::checkTimeout(std::optional<std::chrono::duration<long, std::ratio<1l, 1000l> > >) + 0x1e6 (0x7fe3dcc0c6e6 in /home/qyzheng/anaconda3/envs/isaacgym/lib/python3.8/site-packages/torch/lib/libtorch_cuda.so)
[... cut ...]
frame #3: c10d::ProcessGroupNCCL::ncclCommWatchdog() + 0x119 (0x7fd718c10839 in /home/qyzheng/anaconda3/envs/isaacgym/lib/python3.8/site-packages/torch/lib/libtorch_cuda.so)
frame #4: <unknown function> + 0xdbbf4 (0x7fd8130c7bf4 in /home/qyzheng/anaconda3/envs/isaacgym/bin/../lib/libstdc++.so.6)
frame #5: <unknown function> + 0x94b43 (0x7fd8191cbb43 in /lib/x86_64-linux-gnu/libc.so.6)
frame #6: <unknown function> + 0x126a00 (0x7fd81925da00 in /lib/x86_64-linux-gnu/libc.so.6)

Aborted (core dumped)
```

bowieshi, 2024-11-23:
> If you run with 1 GPU does it work ok? I'd expect NCCL errors only to occur in multi GPU settings.
> 
> We trained on 16 GPUs for a week. With terrains and SAMP. I wouldn't expect linear slowdown with less GPUs. You should see reasonable results within a day, then it takes a while to master the hard motions.

@tesslerc When run with 1 GPU, it is OK. You can see the following reference log. When trained with 1 GPU, it can proceed to  "training" and "collecting data" iteration.
[image]

tesslerc, 2024-11-23:
Ok. I think it might be due to manually running, as opposed to how we run on a cluster on our end.

Our code expects reach GPU to have it's own process. SLURM SRUN does this for us. You should be able to achieve the same outcome with the torchrun command.

https://pytorch.org/docs/stable/elastic/run.html

Could you check if that works? I'll update the readme with the proper instructions if it does.

bowieshi, 2024-11-24:
> Ok. I think it might be due to manually running, as opposed to how we run on a cluster on our end.
> 
> Our code expects reach GPU to have it's own process. SLURM SRUN does this for us. You should be able to achieve the same outcome with the torchrun command (nproc per node).
> 
> https://pytorch.org/docs/stable/elastic/run.html
> 
> Could you check if that works? I'll update the readme with the proper instructions if it does.

Thank you for your reply! After detailed debugging, I found that the problem was caused by a faulty GPU. Whenever I include this specific GPU in my parallel training, my code will be stuck. But when I used another GPU to train, no problems  happen again. Directly using `CUDA_VISIBLE_DEVICES=0,1,4,5 python phys_anim/train_agent.py` can start the training. Sorry for the confusion caused.

closed 2024-11-24 by bowieshi