NVlabs/ProtoMotions #31: Hydra Configuration Error with `motion_file`

Rowing0914, 2024-11-25:
#### **Description**  
When running the command to evaluate an agent with Hydra, the script fails to recognize `motion_file` as part of the configuration, even though it is defined in the YAML file ( `/phys_anim/config/base/structure.yaml` ) as required (`???`). The error suggests the key is not in the configuration structure.

#### **Steps to Reproduce**  
** I'm done with IsaacGym route of installation!

1. Run the following command:  
   ```bash
   python phys_anim/eval_agent.py +robot=smpl +backbone=isaacgym \
   +motion_file=/root/workspace/PULSE-modified/data/amass/amass_test.pkl \
   +checkpoint=./data/pretrained_models/MaskedMimic/last.ckpt
   ```
2. Observe the error:
   ```
   Key 'motion_file' is not in struct
   full_key: motion_file
   object_type=dict
   ```

#### **Environment**  
- Python version: 3.8 .19 
- hydra-core==1.3.2
- Operating System: Linux  

#### Error Log
```
Importing module 'gym_38' (/root/workspace/isaac-gym2/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /root/workspace/isaac-gym2/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
2024-11-25 13:04:56,573 - INFO - logger - logger initialized
phys_anim/eval_agent.py:38: UserWarning: 
The version_base parameter is not specified.
Please specify a compatability version level, or None.
Will assume defaults for version 1.1
  @hydra.main(config_path="config")
Traceback (most recent call last):
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/hydra/_internal/config_loader_impl.py", line 390, in _apply_overrides_to_config
    OmegaConf.update(cfg, key, value, merge=True)
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/omegaconf/omegaconf.py", line 741, in update
    root.__setattr__(last_key, value)
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/omegaconf/dictconfig.py", line 337, in __setattr__
    raise e
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/omegaconf/dictconfig.py", line 334, in __setattr__
    self.__set_impl(key, value)
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/omegaconf/dictconfig.py", line 318, in __set_impl
    self._set_item_impl(key, value)
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/omegaconf/basecontainer.py", line 549, in _set_item_impl
    self._validate_set(key, value)
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/omegaconf/dictconfig.py", line 180, in _validate_set
    target = self._get_node(key) if key is not None else self
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/omegaconf/dictconfig.py", line 475, in _get_node
    self._validate_get(key)
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/omegaconf/dictconfig.py", line 164, in _validate_get
    self._format_and_raise(
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/omegaconf/base.py", line 231, in _format_and_raise
    format_and_raise(
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/omegaconf/_utils.py", line 899, in format_and_raise
    _raise(ex, cause)
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/omegaconf/_utils.py", line 797, in _raise
    raise ex.with_traceback(sys.exc_info()[2])  # set env var OC_CAUSE=1 for full trace
omegaconf.errors.ConfigAttributeError: Key 'motion_file' is not in struct
    full_key: motion_file
    object_type=dict

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "phys_anim/eval_agent.py", line 99, in <module>
    main()
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/hydra/main.py", line 94, in decorated_main
    _run_hydra(
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/hydra/_internal/utils.py", line 394, in _run_hydra
    _run_app(
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/hydra/_internal/utils.py", line 457, in _run_app
    run_and_report(
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/hydra/_internal/utils.py", line 223, in run_and_report
    raise ex
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/hydra/_internal/utils.py", line 220, in run_and_report
    return func()
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/hydra/_internal/utils.py", line 458, in <lambda>
    lambda: hydra.run(
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/hydra/_internal/hydra.py", line 105, in run
    cfg = self.compose_config(
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/hydra/_internal/hydra.py", line 594, in compose_config
    cfg = self.config_loader.load_configuration(
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/hydra/_internal/config_loader_impl.py", line 142, in load_configuration
    return self._load_configuration_impl(
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/hydra/_internal/config_loader_impl.py", line 276, in _load_configuration_impl
    ConfigLoaderImpl._apply_overrides_to_config(config_overrides, cfg)
  File "/root/.conda/envs/maskedmimic/lib/python3.8/site-packages/hydra/_internal/config_loader_impl.py", line 392, in _apply_overrides_to_config
    raise ConfigCompositionException(
hydra.errors.ConfigCompositionException: Could not override 'motion_file'.
To append to your config use +motion_file=/root/workspace/PULSE-modified/data/amass/amass_test.pkl
```

tesslerc, 2024-11-25:
That change shouldn't be needed. The config is loaded from the checkpoints folder. Any command-like configs will override the defaults stored in the model config (which was created with the model).
The error suggests you might be missing a `+` before the `motion_file`. E.g., `+motion_file=..`?

Rowing0914, 2024-11-26:
thank you for your quick reply! 
yeah, I'd tried and got a similar error so I gave up digging in. I think the source of error is that the base config name is now base.yaml, right? so that's why we needed to specify it when launch with hydra?

Anyway, now it's working so I can close this issue!

closed 2024-11-26 by Rowing0914

Rowing0914, 2024-11-26:
I see, that's true... After some trials, the following cmd seems to be running through.

```
CUDA_VISIBLE_DEVICES=0 python phys_anim/eval_agent.py +robot=smpl +backbone=isaacgym +opt=record_video \
+motion_file=phys_anim/data/motions/smpl_humanoid_walk.npy \
+num_envs=4 +experiment_name=test \
+checkpoint=./data/pretrained_models/MaskedMimic/last.ckpt
```

Rowing0914, 2024-11-26:
@tesslerc Thank you for your support!