NVlabs/ProtoMotions #33: Reward and loss curve for AMP isaacsim humanoid_walk incorrect

mulplue, 2024-11-26:
Hi! I trained AMP humanoid walk task using the following script:
`python phys_anim/train_agent.py +exp=amp motion_file=phys_anim/data/motions/amp_humanoid_walk.npy +robot=amp +backbone=isaacsim +opt=wdb`
But the reward and loss curve seems incorrect:
[image]
[image]

Since the task reward keeps 0, what may cause this error? Thanks!

tesslerc, 2024-11-26:
When training AMP there are two reward terms.
One is the discriminator reward, which controls the style. The other is the task reward.

If you train pure AMP, there won't be a task reward. Tasks such as steering and path following introduce a task observation and reward.

mulplue, 2024-11-26:
> When training AMP there are two reward terms. One is the discriminator reward, which controls the style. The other is the task reward.
> 
> If you train pure AMP, there won't be a task reward. Tasks such as steering and path following introduce a task observation and reward.

Thanks! I misinterpreted the 'humanoid walk' as a velocity-following task. I understand now, thanks again!

mulplue, 2024-11-26:
btw, when I try steering/path_following using: 
`python phys_anim/train_agent.py +exp=path_follower motion_file=phys_anim/data/motions/amp_humanoid_walk.npy +robot=amp +backbone=isaacsim +opt=wdb`
output error:
```
Error executing job with overrides: ['+exp=path_follower_humanoid_walk', '+robot=amp', '+backbone=isaacsim', '+opt=wdb']
Error in call to target 'phys_anim.envs.path_following.isaacsim.PathFollowingHumanoid':
TypeError("PathFollowingHumanoid.set_up_scene() missing 1 required positional argument: 'scene'")
```
Is this part not ready yet? or just a bug

tesslerc, 2024-11-26:
Thanks for catching this.
Indeed IsaacSim pathfollower/steering hasn't been tested for a while, so that part of the integration seems to be broken.
I believe removing `scene` from the inputs to `def set_up_scene` and everywhere it passes it on to, including from `def _load_marker_asset` should fix it.

IsaacLab provides a nicer interface on-top of IsaacSim, so I hope to release an update to the code in the next week or two that shifts from Sim to Lab. It should fix these issues and also make it much easier to work with.

dbdxnuliba, 2024-12-02:
Is it that the closer the discriminators_reward is to 0, the more the amp style learns to resemble it, or is it that the closer the discriminators_reward approaches 1, the more the amp style learns to resemble it?

[image]

tesslerc, 2024-12-02:
Higher discriminator reward means the discriminator isn't able to distinguish between the demonstration data and the simulated motion.

That typically means the motion is more similar to the data.

dbdxnuliba, 2024-12-03:
[image]

ok, I will have a try to train h1 on amp algo,

tesslerc, 2024-12-16:
@mulplue , thanks for the heads up.
Indeed path follower and steering weren't supported yet in IsaacLab.
It's now updated.

(the visualization marker for steering needs fixing, but it should give a general idea for now)

mulplue, 2024-12-17:
> @mulplue , thanks for the heads up. Indeed path follower and steering weren't supported yet in IsaacLab. It's now updated.

Thanks for your reply! Will you release this update anytime soon?

tesslerc, 2024-12-17:
> > @mulplue , thanks for the heads up. Indeed path follower and steering weren't supported yet in IsaacLab. It's now updated.
> 
> Thanks for your reply! Will you release this update anytime soon?

Yes, as I wrote in the previous message, the code has been updated.
Please check the latest version and let me know if you encounter any issues. 🙏🏼

mulplue, 2024-12-17:
> Yes, as I wrote in the previous message, the code has been updated. Please check the latest version and let me know if you encounter any issues. 🙏🏼

Yes! I can successfully run it now, just encountered some cuda errors because of the limited memory(I use 4090 24G), which can be fixed by setting a lower num_envs, post here for reference:
```
2024-12-17 09:04:18 [133,957ms] [Error] [omni.physx.plugin] Cuda context manager error, simulation will be stopped and new cuda context manager will be created.
2024-12-17 09:04:18 [133,957ms] [Error] [omni.physx.plugin] PhysX error: GPU compressContactStage2 fail to launch kernel stage 1!!
, FILE /builds/omniverse/physics/physx/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp, LINE 9607
```
I'll run the experiments to check if the results align with the original amp paper. I'll close this issue if it works, thanks!

mulplue, 2024-12-19:
@tesslerc Hi, I found that the "config/env/amp.yaml" was lost after the **breaking update** commit, while the "path_follower" task still needs `default: -amp`(in "config/env/path_follower.yaml"). 
Should I use the latest amp.yaml before the **breaking update**? or just replace the `default: -amp` with `default -humanoid`.

tesslerc, 2024-12-19:
Thank you for testing this!
Your insights are correct on both things.

1. The default env should change to humanoid. We no long need a specialized AMP environment.
2. The AMP environment set the discriminator to enabled by default, this was reverted when removing the specialized env, which is why things broke.

I believe [this](https://github.com/NVlabs/ProtoMotions/commit/d98c39b4ca816cbeb0efe21d557c11f219674f1b) fix should solve your issues.
Could you test this and let me know if it is now solved?

Also, keep in mind that IsaacLab has an updated mesh solver. This leads to some failure to "cook" large terrains.
From my experience using num_terrains set to 10 fails and the character falls through the ground. 7 usually works well.
IsaacLab throws a warning when this happens.

mulplue, 2024-12-19:
> Thank you for testing this! Your insights are correct on both things.
>
> 1. The default env should change to humanoid. We no long need a specialized AMP environment.
> 2. The AMP environment set the [...]

Thanks! I'll test this new config.
I also found the issue of "the robot falls through the ground" in the visualization of the "path_follower" task. I thought it was a problem with my running environment at first. This is really helpful! I'll try it out and visualize it before training.

mulplue, 2024-12-20:
@tesslerc Hi, I've tested the new config, and I found that **"the discriminator reward of isaaclab may be incorrect".**
[image]
The discriminator reward of isaaclab seems hard to jump over 0. The visualization shows that:
1. in isaaclab, the robot finishes the task in strange motion(with ~0 discriminator reward)
[image]
2. in isaacgym, the motion of the robot seems good compared to isaaclab(I think it could be better when steps increase)
[image]

Since both backbones share the same env/algo/... configs, it seems strange why this happens. Could it be some problems with the physics solver of isaaclab?

tesslerc, 2024-12-21:
I'll train some AMP agents in IsaacLab and test this.
Just to make sure, both the IsaacLab and IsaacGym examples you shared above -- both are from the same commit version, is this correct?

tesslerc, 2024-12-22:
@mulplue , there was an issue in the IsaacLab code that prevented proper initialization and update of the AMP history buffer.
This was likely introduced during the transition from IsaacSim to IsaacLab.

It has now been fixed. I have verified that indeed before the fix the discriminator rewards were very low (near 0) and are now more similar to what you observe in IsaacGym (in the range of 0.1 to 0.6).

mulplue, 2024-12-23:
> @mulplue , there was an issue in the IsaacLab code that prevented proper initialization and update of the AMP history buffer. This was likely introduced during the transition from IsaacSim to [...]

Sry I didn't check my email over the weekend. Yes I use the same version, thanks for this fix! I'll try this version.

mulplue, 2024-12-23:
@tesslerc I found some small variable renaming bugs in **envs/humanoid** "common.py" and "isaaclab.py" and made a pull request for this, see https://github.com/NVlabs/ProtoMotions/pull/42
Also, when I test the gym version, I found that the `self.reset_states` and `self.object_reset_states` will be reset to `None` every `self.pre_physics_step()` thus making  ```TypeError: 'NoneType' object is not subscriptable``` error in `self.set_env_state()`. I think the reset function may need to be reorganized(so I post it here instead of a PR)

tesslerc, 2024-12-23:
Thanks. My bad for only checking on the internal codebase. I've accepted the PR and fixed the other issue. Verified it runs on the public codebase.

Thanks a lot for helping test and provide feedback!

mulplue, 2024-12-25:
Hi @tesslerc , the new reward curves are still incorrect:
[image]

some observations:
1. (the main thing)the discriminator reward is too low, the robot did not learn any style(double-checked in visualization)
2. the "steering" in isaac lab(red curve) may met some bugs? I'm not sure whether it's a bug of isaac lab

mulplue, 2025-01-15:
@tesslerc Thank you for your response! To verify AMP's implementation, I conducted testing by migrating the 'sword and shield' robot and its motions from the ASE repo(https://github.com/nv-tlabs/ASE). Here are the **3 experiments I tested**:
1. Test pure motion imitation using AMP to replicate a strike motion, without any specific task
2. Test path_follower and steering tasks using robot='sword and shield' and motion_file=<ase_motion_file>
3. Migration of several tasks from ASE (location, heading, reach) and testing them with robot='sword and shield' and motion_file=<ase_motion_file> to verify alignment with ASE results

**Results:**
1. Success, as shown in video 1
2. Success, though the movements might appear unusual due to the inclusion of strike motions in the ASE motion_file, see video 2 (path_follower)
3. Only the heading task was successful. For the other tasks (location, heading), the converged task reward was low while the discriminator reward was high. In ASE, I observed that task reward initially increases from 0 to a high value (with disc reward near 0), followed by an increase in disc reward. However, in ProtoMotions, disc reward starts high and continues to increase, while task reward increases slowly and converges to a low value. I compared amp&ppo parameters between ASE and ProtoMotions but found no obvious differences. I wonder if you have any insights into this matter.

In conclusion, I believe the AMP implementation is correct, though there might be some minor parameter inconsistencies with ASE's implementation. I will close this issue. Thanks!

https://github.com/user-attachments/assets/538b879d-3230-4063-88a1-140864010695
https://github.com/user-attachments/assets/e6685dcf-f65f-4bf2-aa86-7152e18ab00e

closed 2025-01-15 by mulplue