NVlabs/ProtoMotions #35: How to Run Evaluation for Text Control Using the Pretrained Model?

Rowing0914, 2024-11-28:
Hi! Could you tell me how to run evaluation for text control using the pretrained model?

Thanks!

tesslerc, 2024-11-28:
For text control you should run
```
python phys_anim/eval_agent.py +robot=smpl +backbone=isaacgym \
+motion_file=phys_anim/data/motions/smpl_humanoid_walk.npy \
+checkpoint=./data/pretrained_models/MaskedMimic/last.ckpt \
+opt=masked_mimic/tasks/user_control
```

When you press `U` it will create a temporary `.YAML` file in `/tmp/` which you can edit.
Set `text_control: True`, define a text prompt, save the file, and then press `U` again.

Rowing0914, 2024-11-29:
Here is another issue just came up!
I'm working on a headless machine without access to a GPU desktop. I've attached a video demonstrating the use of your command with a headless rendering tool ([Web Viewer of SKRL](https://skrl.readthedocs.io/en/1.1.0/api/utils/isaacgym_utils.html#web-viewer-for-development-without-x-server)) to print [inference_parameters["text_command"]](https://github.com/NVlabs/ProtoMotions/blob/main/phys_anim/envs/masked_mimic/tasks/user_control/common.py#L100) and its visibility per timestep.

Since the web viewer doesn't support mouse or keyboard input, I added update_inference_parameters right [after the environment instantiation](https://github.com/NVlabs/ProtoMotions/blob/main/phys_anim/envs/masked_mimic/tasks/user_control/common.py#L49).

Based on the printed values in the video, the text command and its visibility seem fine, but the agent doesn't respond to the command. Below is my YAML configuration, which I believe is correct:

```yaml
constrained_joints:
- body_name: L_Ankle
  constraint_state: 1
- body_name: R_Ankle
  constraint_state: 1
- body_name: Pelvis
  constraint_state: 1
- body_name: Head
  constraint_state: 1
- body_name: L_Hand
  constraint_state: 1
- body_name: R_Hand
  constraint_state: 1
inbetweening: false
motion_id: 0
object_conditioning: false
text_command: a person is jumping while putting arms in the air.
text_conditioned: True
```

[Video attachment link](https://github.com/user-attachments/assets/e94441ca-2fd2-428a-a83b-da95daea7489)

Could you help me figure out why the agent isn't following the text prompt?

Thank you so much for your support!

tesslerc, 2024-11-29:
Does this command work in the standard flow? With keyboard command.

Reason I'm asking, and this relates to your other question, is that text works to a certain extent but it's far from "production ready".
That's why we didn't compare text in the paper and didn't build a text evaluation pipeline yet. It felt it's a nice feature that sometimes works and can be useful. It's nice to have, but it's not yet on par with what you get from text-only models.

tesslerc, 2024-12-02:
Isaac has a live streaming mode. It enables rendering on a remote PC with a GPU and only viewing (streaming) locally.

I haven't implemented that feature yet, but if you do and find it useful I'd be very happy to accept a PR.

Rowing0914, 2024-12-02:
oh, you mean for Issac-Sim?

Rowing0914, 2024-12-02:
I’m having trouble locating the specific line where the text condition is fed to the agent. For instance, where the text-cmd-emb is added to the observation or any other mechanism used to incorporate text conditions into the agent’s inputs. Could you point me to the relevant line or section?

Rowing0914, 2024-12-02:
It looks like the issue with the agent's insensitivity to the text-cmd arises from the missing text-embedding in the input keys [[here](https://github.com/NVlabs/ProtoMotions/blob/main/phys_anim/agents/models/mlp.py#L81)](https://github.com/NVlabs/ProtoMotions/blob/main/phys_anim/agents/models/mlp.py#L81). The text-embedding can be set from [[this yaml file](https://github.com/NVlabs/ProtoMotions/blob/main/data/pretrained_models/MaskedMimic/config.yaml#L108)](https://github.com/NVlabs/ProtoMotions/blob/main/data/pretrained_models/MaskedMimic/config.yaml#L108) in the pretrained model's directory.

However, I'm now encountering an error when loading the pretrained model:

```shell
RuntimeError: Error(s) in loading state_dict for ActorFixedSigmaVAE:
        Missing key(s) in state_dict: "vae_encoder.trunk.extra_input_models.motion_text_embeddings.running_obs_norm.mean", "vae_encoder.trunk.extra_input_models.motion_text_embeddings.running_obs_norm.var", "vae_encoder.trunk.extra_input_models.motion_text_embeddings.running_obs_norm.count", "vae_encoder.trunk.extra_input_models.motion_text_embeddings.trunk.0.weight", "vae_encoder.trunk.extra_input_models.motion_text_embeddings.trunk.0.bias", "vae_encoder.trunk.extra_input_models.motion_text_embeddings.trunk.2.weight", "vae_encoder.trunk.extra_input_models.motion_text_embeddings.trunk.2.bias", "vae_encoder.trunk.extra_input_models.motion_text_embeddings.trunk.4.weight", "vae_encoder.trunk.extra_input_models.motion_text_embeddings.trunk.4.bias". 
        size mismatch for vae_encoder.trunk.trunk.0.weight: copying a param with shape torch.Size([1024, 5074]) from checkpoint, the shape in current model is torch.Size([1024, 5586]).
```

Do you know how to resolve this?

tesslerc, 2024-12-02:
We don't use the encoder during inference. You want to look at the prior.
See here: https://github.com/NVlabs/ProtoMotions/blob/ed21d6339162c3bbf406afe32cd72873c37c5446/data/pretrained_models/MaskedMimic/config.yaml#L159

I would try the following:
1. Set the constrained joints to null. This makes sure it receives no joint position/rotation constraints.
```
constrained_joints: null
```
2. Try simpler text prompts, just to see if it's connected. For example, "a person sits down" or "a person is turning around in place". I recall that these typically work well.

Rowing0914, 2024-12-06:
Hi @tesslerc ,

Thank you so much for your support and the insightful chat at SIGGRAPH Asia! Your solution worked perfectly—now the agent sits down as intended! 🎉
- Here’s a short [video demonstration](https://youtu.be/QeUE5DZBtI0?si=zF2m6e989HIVoXyP&t=21).

I’ll continue exploring further!

Here’s the YAML configuration I used, in case it’s helpful for someone in the future:

```yaml
constrained_joints: null
inbetweening: false
motion_id: 0
object_conditioning: false
text_command: a person sits down
text_conditioned: true
```

Thanks again for your help!

tesslerc, 2024-12-06:
Amazing! Thanks for sharing and really happy it's working for you!

I think I should add an option for longer motion generation when it's not in "tracking mode". 
Are there additional features you think might be useful? (Within masked mimics capability envelope).

Rowing0914, 2024-12-08:
Added the support of multiple agents spawning!
- https://youtu.be/p6MUSYQAEm8

```yaml: config.yaml
constrained_joints: null
inbetweening: false
motion_id: 0
object_conditioning: false
text_command:
  - a person sits down, then jump forward.
  - a person kicking forward.
  - a person standing on one leg trying to balance himself.
text_conditioned: True

```

closed 2024-12-17 by tesslerc