NVlabs/ProtoMotions #36: can't find pkl file but only npz files in smplh

dbdxnuliba, 2024-12-02:
~/rl/ProtoMotions$ python data/scripts/convert_amass_to_isaac.py /home/era/rl/ProtoMotions/data/AMASS_Complete/ --humanoid-type=smplh

[image]

but I can't find pkl file but only npz files in smplh
https://mano.is.tue.mpg.de/download.php
[image]
[image]

tesslerc, 2024-12-02:
I downloaded the SMPL-X files from https://smpl-x.is.tue.mpg.de/ and use the flag `--humanoid-type=smplx`

dbdxnuliba, 2024-12-03:
thanks ,I see

closed 2024-12-03 by dbdxnuliba