NVlabs/ProtoMotions #37: errors with amp

BRoln7, 2024-12-04:
I used the SMPL humanoid to train an amp task, and I found in inference stage, the humanoid just stand, what is the problem? The motion file contains several actions including walking and running

tesslerc, 2024-12-04:
What command did you run for training and what for test?

`phys_anim/train_agent.py +exp=amp +robot=smpl +backbone=isaacgym motion_file=phys_anim/data/motions/smpl_humanoid_walk.npy` should result in a simple walking motion with less than an hour.

BRoln7, 2024-12-04:
OK，i'll check later

发送自 Outlook for iOS<https://aka.ms/o0ukef>

BRoln7, 2024-12-10:
How should I prepare a dataset when training for a path-following task? Specifically, how should I select motions from the AMASS dataset?

tesslerc, 2024-12-10:
Convert the dataset to Isaac format, see the data section in the README.
Then, create a `.yaml` file containing the motion styles you want as a prior for your policy. See examples in the `data/yaml_files/` folder for how these files should be structured.
Then, as explained in the README, you can pre-package the data for faster loading.

The result is a `.pt` file that you provide to the flag `motion_file=<path to .pt file>`.

BRoln7, 2024-12-10:
Thank you for your reply! But actually, I would like to know how to determine which motions from the AMASS dataset are what I need, since the dataset is quite large.

tesslerc, 2024-12-10:
In [this](https://github.com/NVlabs/ProtoMotions/blob/main/data/yaml_files/amass_train.yaml) file you have all the motions and their textual labels.
For path following you can focus on walk/run/crouch-walk/crawl/stand as a good representation of the types of motions.

For reproducing the Trace and Pace paper, I suggest looking into the motions they used [here](https://github.com/nv-tlabs/pacer).

tesslerc, 2024-12-10:
The video doesn't load.
How many motions did you include? Did you turn off the height conditioning in the path follower config?

BRoln7, 2024-12-11:
Yes, I turned it off. I included five walking motions from KIT in amass dataset.

tesslerc, 2024-12-11:
Can you try adding some more motions and see if that's still the case?

BRoln7, 2024-12-20:
When I load motions, it ocurrs such a problem:
```
/home/bingyi/miniconda3/envs/isaaclab/lib/python3.10/site-packages/smplx/body_models.py:137: DeprecationWarning: Please use `csc_matrix` from the `scipy.sparse` namespace, the `scipy.sparse.csc` namespace is deprecated.
  data_struct = Struct(**pickle.load(smpl_file,
/home/bingyi/miniconda3/envs/isaaclab/lib/python3.10/site-packages/chumpy/ch.py:20: DeprecationWarning: Please use `LinearOperator` from the `scipy.sparse.linalg` namespace, the `scipy.sparse.linalg.interface` namespace is deprecated.
  from scipy.sparse.linalg.interface import LinearOperator
/home/bingyi/miniconda3/envs/isaaclab/lib/python3.10/site-packages/chumpy/ch.py:1203: DeprecationWarning: inspect.getargspec() is deprecated since Python 3.0, use inspect.signature() or inspect.getfullargspec()
  want_out = 'out' in inspect.getargspec(func).args
Sampling motion: tensor([0, 0, 0, 0])
/home/bingyi/socnav_sim/poselib/poselib/skeleton/skeleton3d.py:1240: DeprecationWarning: Please use `gaussian_filter1d` from the `scipy.ndimage` namespace, the `scipy.ndimage.filters` namespace is deprecated.
  filters.gaussian_filter1d(velocity, 2, axis=-3, mode="nearest")
  0%|                                                                                    | 0/3 [00:00<?, ?it/s]
```
Then the progress of the load action is stuck.I wonder what the problem is？

tesslerc, 2024-12-22:
I haven't encountered such an issue before.
Did you figure out why this happens?

BRoln7, 2024-12-23:
I haven't solved it yet, but I'll try it later. I'll reply when I'm done

BRoln7, 2024-12-24:
There's another question I need to trouble you with: How can I apply meshes to the character in Isaac Lab? Maybe like this:
[image]

tesslerc, 2024-12-24:
I'd suggest asking in the official IsaacLab/IsaacSim forums. You should find the relevant experts over there.

closed 2025-03-23 by tesslerc