NVlabs/ProtoMotions #40: Could you provide the learning curves for reference?

CavalloneChen, 2024-12-12:
Hi, thank you for open sourcing such a wonderful work.

I find the training to converge pretty slowly (stage 1). I wonder could you provide a learning curve plot for the whole amassx dataset training? I'm concerned something may have gone wrong on my side.

tesslerc, 2024-12-17:
[image]
Here's an example of the tracking success rate. This is without the objects, only terrains, but with SMPL-X humanoid.
After ~6500 epochs it reaches 99% success rate on the training set. From there it's a slow grind upwards.

CavalloneChen, 2024-12-19:
> ![image](https://private-user-images.githubusercontent.com/3133866/396390603-0fbb90b6-8174-444b-ae23-0865df665c4c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3L [...]

Thank you for your reply!
Is this curve reporting the tracking success rate for a **single motion clip** or the **whole amass dataset**? And I assume the training budget is the same as indicated in the MaskedMimic paper (num_envs=16384, 4 A100 GPU) ?

tesslerc, 2024-12-19:
The entire amass dataset.
When training on a single example, does it converge successfully and in reasonable time?

bowieshi, 2025-02-26:
> Thank you for your reply! Is this curve reporting the tracking success rate for a **single motion clip** or the **whole amass dataset**? And I assume the training budget is the same as indicated in [...]

Hi, can I ask that did you successfully train the code with 16384 num_env in A100? I met some memory error with my 80GB A100. I can only run with 1024 env. With larger env, my training would crash after serveral training epochs.

minolta001, 2025-07-02:
Hi! May I know your hardware configuration? I am using a single 3090. 

However, instead of training on whole AMASS, I only use CMU 07 dataset from AMASS. Since you mentioned the same issue, I feel curious about the impact from the size of dataset. I may plan to start training with 2 or 3 motions and see if the learning can be done.

tesslerc, 2025-07-06:
@minolta001 , the G1 has 23 actions whereas SMPL is at 69. I'm not sure I agree with the larger action space.
I hope we can release some updates from our side fairly soon, which hopefully should help even further.