NVlabs/ProtoMotions #43: Example Amp Training Error

Steven9Smith, 2024-12-29:
When I run
../IsaacLab/isaaclab.sh -p phys_anim/train_agent.py +exp=amp motion_file=data/motions/amp_humanoid_walk.npy +robot=amp +backbone=isaaclab headless=true

I get the error output:

both 

../IsaacLab/isaaclab.sh -p phys_anim/train_agent.py +exp=amp motion_file=data/motions/amp_humanoid_walk.npy +robot=amp +backbone=isaaclab headless=true env=humanoid
[INFO] Using python from: /media/linux-test/BigMomma/anaconda/envs/isaacsim2/bin/python
wandb: WARNING This integration is tested and supported for lightning Fabric 2.1.3.
wandb: WARNING             Please report any issues to https://github.com/wandb/wandb/issues with the tag `lightning-fabric`.
2024-12-29 15:31:53,538 - INFO - logger - logger initialized
/media/linux-test/BigMomma/BMPublic/Github/Nvidia/ProtoMotions/phys_anim/train_agent.py:67: UserWarning:
The version_base parameter is not specified.
Please specify a compatability version level, or None.
Will assume defaults for version 1.1
  @hydra.main(config_path="config", config_name="base")
/media/linux-test/BigMomma/anaconda/envs/isaacsim2/lib/python3.10/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'base': Defaults list is missing `_self_`. See https://hydra.cc/docs/1.2/upgrades/1.0_to_1.1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
anaconda/envs/isaacsim2/lib/python3.10/site-packages/hydra/_internal/hydra.py:119: UserWarning: Future Hydra versions will no longer change working directory at job runtime by default.
See https://hydra.cc/docs/1.2/upgrades/1.1_to_1.2/changes_to_job_working_dir/ for more information.
  ret = run_job(
Initializing distributed: GLOBAL_RANK: 0, MEMBER: 1/1
----------------------------------------------------------------------------------------------------
distributed_backend=nccl
All distributed processes registered. Starting with 1 processes
----------------------------------------------------------------------------------------------------

[rank: 0] Seed set to 0
Setting seed: 0
[INFO][AppLauncher]: Loading experience file: BMPublic/Github/Nvidia/IsaacLab/source/apps/isaaclab.python.headless.kit
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Loading user config located at: 'anaconda/envs/isaacsim2/lib/python3.10/site-packages/omni/data/Kit/Isaac-Sim/4.2/user.config.json'
[Info] [carb] Logging to file: anaconda/envs/isaacsim2/lib/python3.10/site-packages/omni/logs/Kit/Isaac-Sim/4.2/kit_20241229_153154.log
2024-12-29 20:31:54 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[2024-12-29 15:31:54,269][asyncio][DEBUG] - Using selector: EpollSelector
[2024-12-29 15:31:54,412][omni.kit.telemetry.impl.sentry_extension][INFO] - sentry is disabled for external build
[2024-12-29 15:31:54,412][omni.kit.telemetry.impl.sentry_extension][INFO] - sentry is disabled for external build
2024-12-29 20:31:54 [231ms] [Warning] [omni.platforminfo.plugin] failed to open the default display.  Can't verify X Server version.

|---------------------------------------------------------------------------------------------|
| Driver Version: 560.35.05     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 3090          | Yes: 0 |     | 24822   MB | 10de      | 0          |
|     |                                  |        |     |            | 2204      | 225bd100.. |
|     |                                  |        |     |            | 9         |            |
|---------------------------------------------------------------------------------------------|
| 1   | NVIDIA GeForce RTX 3090          | Yes: 1 |     | 24822   MB | 10de      | 0          |
|     |                                  |        |     |            | 2204      | 132dd30c.. |
|     |                                  |        |     |            | a         |            |
|=============================================================================================|
| OS: 22.04.5 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.5, Kernel: 6.8.0-50-generic
| Processor: AMD Ryzen 9 5950X 16-Core Processor | Cores: 16 | Logical: 32
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 128716 | Free Memory: 121975
| Total Page/Swap (MB): 2047 | Free Page/Swap: 2047
|---------------------------------------------------------------------------------------------|
anaconda/envs/isaacsim2/lib/python3.10/site-packages/torch/functional.py:513: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at ../aten/src/ATen/native/TensorShape.cpp:3609.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
Error executing job with overrides: ['+exp=amp', 'motion_file=data/motions/amp_humanoid_walk.npy', '+robot=amp', '+backbone=isaaclab', 'headless=true', 'env=humanoid']
[rank0]: Traceback (most recent call last):
[rank0]:   File "anaconda/envs/isaacsim2/lib/python3.10/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 92, in _call_target
[rank0]:     return _target_(*args, **kwargs)
[rank0]:   File "Github/Nvidia/ProtoMotions/phys_anim/envs/humanoid/isaaclab.py", line 56, in __init__
[rank0]:     super().__init__(config, device, simulation_app)
[... 18 lines cut ...]
[rank0]:     raise ex
[rank0]:   File "anaconda/envs/isaacsim2/lib/python3.10/site-packages/hydra/_internal/utils.py", line 220, in run_and_report
[rank0]:     return func()
[rank0]:   File "anaconda/envs/isaacsim2/lib/python3.10/site-packages/hydra/_internal/utils.py", line 458, in <lambda>
[rank0]:     lambda: hydra.run(
[rank0]:   File "anaconda/envs/isaacsim2/lib/python3.10/site-packages/hydra/_internal/hydra.py", line 132, in run
[rank0]:     _ = ret.return_value
[rank0]:   File "anaconda/envs/isaacsim2/lib/python3.10/site-packages/hydra/core/utils.py", line 260, in return_value
[rank0]:     raise self._return_value
[rank0]:   File "anaconda/envs/isaacsim2/lib/python3.10/site-packages/hydra/core/utils.py", line 186, in run_job
[rank0]:     ret.return_value = task_function(task_cfg)
[rank0]:   File "Github/Nvidia/ProtoMotions/phys_anim/train_agent.py", line 115, in main
[rank0]:     env = instantiate(
[rank0]:   File "anaconda/envs/isaacsim2/lib/python3.10/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 227, in instantiate
[rank0]:     return instantiate_node(
[rank0]:   File "anaconda/envs/isaacsim2/lib/python3.10/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 348, in instantiate_node
[rank0]:     return _call_target(_target_, partial, args, kwargs, full_key)
[rank0]:   File "anaconda/envs/isaacsim2/lib/python3.10/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 98, in _call_target
[rank0]:     raise InstantiationException(msg) from e
[rank0]: hydra.errors.InstantiationException: Error in call to target 'phys_anim.envs.humanoid.isaaclab.Humanoid':
[rank0]: ValueError('Unsupported robot type: amp_humanoid_3d')
[rank0]: full_key: env
[rank0]: ()

[rank0]: kwargs = {'config': {'experiment_name': 'amp', 'num_envs': 4096, 'headless': True, 'backbone': 'isaaclab', 'max_episode_length': 300, 'robot': {'dof_obs_size': 72, 'number_of_actions': 36, 'self_obs_size': 223, 'self_obs_max_coords_size': 223, 'num_bodies': 15, 'contact_bodies': ['pelvis', 'torso', 'head', 'right_upper_arm', 'right_lower_arm', 'right_hand', 'left_upper_arm', 'left_lower_arm', 'left_hand', 'right_thigh', 'right_shin', 'right_foot', 'left_thigh', 'left_shin', 'left_foot'], 'isaacgym_body_names': ['pelvis', 'torso', 'head', 'right_upper_arm', 'right_lower_arm', 'right_hand', 'left_upper_arm', 'left_lower_arm', 'left_hand', 'right_thigh', 'right_shin', 'right_foot', 'left_thigh', 'left_shin', 'left_foot'], 'isaac [... cut ...] ': 256, 'use_object_identity': False, 'num_point_features': 768, 'num_identity_features': 0, 'num_features_per_object': 843}, 'object_ref_respawn_offset': 0.01, 'visualize_markers': True, 'sync_motion': False, 'output_motion': False, 'camera': None, 'export_video': False, 'export_motion': False, 'record_viewer': False, 'viewer_record_dir': 'output/recordings/viewer'}, 'device': device(type='cuda', index=0), 'simulation_app': <omni.isaac.kit.simulation_app.SimulationApp object at 0x7f2e568eb6d0>}

tesslerc, 2024-12-30:
AMP humanoid is not properly supported in IsaacLab.
In IsaacLab we currently fully support the SMPL and SMPL-X humanoids.

Steven9Smith, 2024-12-30:
I see, took a bit to get isaacgym setup but I can confirm that is was training after setup was complete :)

closed 2024-12-30 by Steven9Smith