NVlabs/ProtoMotions #44: AttributeError: 'MimicHumanoid' object has no attribute self.num_bodies during eval_agent.py execution

tonywang-0517, 2024-12-30:
## Description  
Hi,  
I got an error when running the following command:  
`python phys_anim/eval_agent.py +robot=h1 +backbone=isaacgym +motion_file=data/motions/h1_punch.npy +checkpoint=results/full_body_tracker/lightning_logs/version_0/last.ckpt +headless=False +force_flat_terrain=True`

I got:
```
Loading object assets... ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 100% 0:00:00
=========== Total number of unique objects is 21
Creating 1 environments... ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:--
Error executing job with overrides: ['+robot=h1', '+backbone=isaacgym', '+motion_file=data/motions/h1_punch.npy', '+checkpoint=results/full_body_tracker/lightning_logs/version_0/last.ckpt', '+headless=False', '+force_flat_terrain=True']
[rank0]: Traceback (most recent call last):
[rank0]:   File "/home/tony/miniconda3/envs/isaacgym/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 92, in _call_target
[rank0]:     return _target_(*args, **kwargs)
[rank0]:   File "/home/tony/workspace/ProtoMotions/phys_anim/envs/mimic/isaacgym.py", line 39, in __init__
[rank0]:     super().__init__(config, device)
[rank0]:   File "/home/tony/workspace/ProtoMotions/phys_anim/envs/mimic/common.py", line 52, in __init__
[rank0]:     super().__init__(config, device, *args, **kwargs)
[rank0]:   File "/home/tony/workspace/ProtoMotions/phys_anim/envs/humanoid/isaacgym.py", line 49, in __init__
[rank0]:     super().__init__(config, device)
[rank0]:   File "/home/tony/workspace/ProtoMotions/phys_anim/envs/humanoid/common.py", line 80, in __init__
[rank0]:     super().__init__(config, device, *args, **kwargs)
[rank0]:   File "/home/tony/workspace/ProtoMotions/phys_anim/envs/base_interface/isaacgym.py", line 64, in __init__
[rank0]:     self.create_sim()
[rank0]:   File "/home/tony/workspace/ProtoMotions/phys_anim/envs/humanoid/isaacgym.py", line 226, in create_sim
[rank0]:     self.create_envs(
[rank0]:   File "/home/tony/workspace/ProtoMotions/phys_anim/envs/mimic/isaacgym.py", line 52, in create_envs
[rank0]:     super().create_envs(num_envs, spacing, num_per_row)
[rank0]:   File "/home/tony/workspace/ProtoMotions/phys_anim/envs/humanoid/isaacgym.py", line 353, in create_envs
[rank0]:     self.build_env(i, env_ptr, humanoid_asset)
[rank0]:   File "/home/tony/workspace/ProtoMotions/phys_anim/envs/mimic/isaacgym.py", line 82, in build_env
[rank0]:     self._build_marker(env_id, env_ptr)
[rank0]:   File "/home/tony/workspace/ProtoMotions/phys_anim/envs/mimic/isaacgym.py", line 87, in _build_marker
[rank0]:     num_mimic_tracking_points = self.num_bodies
[rank0]: AttributeError: 'MimicHumanoid' object has no attribute 'num_bodies'

```
I swap the 2 lines https://github.com/NVlabs/ProtoMotions/blob/672d700d5cc3d06b5663043be7f978f322a00f54/phys_anim/envs/humanoid/common.py#L79
from
```
super().__init__(config, device, *args, **kwargs)
self.setup_character_props()
```
 to 

```
self.setup_character_props()
super().__init__(config, device, *args, **kwargs)
```
Then it works

tesslerc, 2024-12-30:
Thanks. There was a mismatch in behavior between IsaacGym and Lab.
Split the `setup_character_props` to handle setup pre-asset loading and post-asset loading.

Pushed an update. Please let me know if you encounter any further issues.

tonywang-0517, 2025-01-11:
Thanks for the updates @tesslerc  
It seems I found another issue when I try to run 
`python phys_anim/eval_agent.py +robot=smpl +backbone=isaacgym +opt=masked_mimic/tasks/story  +checkpoint=data/pretrained_models/MaskedMimic/last.ckpt  +force_flat_terrain=True +motion_file=data/amass/my_amass_test.yaml`

I got:

`Error executing job with overrides: ['+robot=smpl', '+backbone=isaacgym', '+opt=masked_mimic/tasks/story', '+checkpoint=data/pretrained_models/MaskedMimic/last.ckpt', '+force_flat_terrain=True', '+motion_file=data/amass/test.yaml']
[rank0]: Traceback (most recent call last):
[rank0]:   File "phys_anim/eval_agent.py", line 124, in <module>
[rank0]:     main()
[rank0]:   File "/home/tony/miniconda3/envs/isaacgym/lib/python3.8/site-packages/hydra/main.py", line 90, in decorated_main
[rank0]:     _run_hydra(
[rank0]:   File "/home/tony/miniconda3/envs/isaacgym/lib/python3.8/site-packages/hydra/_internal/utils.py", line 389, in _run_hydra
[rank0]:     _run_app(
[rank0]:   File "/home/tony/miniconda3/envs/isaacgym/lib/python3.8/site-packages/hydra/_internal/utils.py", line 452, in _run_app
[rank0]:     run_and_report(
[rank0]:   File "/home/tony/miniconda3/envs/isaacgym/lib/python3.8/site-packages/hydra/_internal/utils.py", line 216, in run_and_report
[rank0]:     raise ex
[rank0]:   File "/home/tony/miniconda3/envs/isaacgym/lib/python3.8/site-packages/hydra/_internal/utils.py", line 213, in run_and_report
[rank0]:     return func()
[rank0]:   File "/home/tony/miniconda3/envs/isaacgym/lib/python3.8/site-packages/hydra/_internal/utils.py", line 453, in <lambda>
[rank0]:     lambda: hydra.run(
[rank0]:   File "/home/tony/miniconda3/envs/isaacgym/lib/python3.8/site-packages/hydra/_internal/hydra.py", line 132, in run
[rank0]:     _ = ret.return_value
[rank0]:   File "/home/tony/miniconda3/envs/isaacgym/lib/python3.8/site-packages/hydra/core/utils.py", line 260, in return_value
[rank0]:     raise self._return_value
[rank0]:   File "/home/tony/miniconda3/envs/isaacgym/lib/python3.8/site-packages/hydra/core/utils.py", line 186, in run_job
[rank0]:     ret.return_value = task_function(task_cfg)
[rank0]:   File "phys_anim/eval_agent.py", line 120, in main
[rank0]:     algo.evaluate_policy()
[rank0]:   File "/home/tony/miniconda3/envs/isaacgym/lib/python3.8/site-packages/torch/utils/_contextlib.py", line 116, in decorate_context
[rank0]:     return func(*args, **kwargs)
[rank0]:   File "/home/tony/workspace/ProtoMotions/phys_anim/agents/ppo.py", line 872, in evaluate_policy
[rank0]:     self.pre_evaluate_policy()
[rank0]:   File "/home/tony/workspace/ProtoMotions/phys_anim/agents/ppo.py", line 900, in pre_evaluate_policy
[rank0]:     self.handle_reset()
[rank0]:   File "/home/tony/workspace/ProtoMotions/phys_anim/agents/mimic_vae.py", line 86, in handle_reset
[rank0]:     super().handle_reset(done_indices)
[rank0]:   File "/home/tony/workspace/ProtoMotions/phys_anim/agents/amp.py", line 238, in handle_reset
[rank0]:     super().handle_reset(done_indices)
[rank0]:   File "/home/tony/workspace/ProtoMotions/phys_anim/agents/ppo.py", line 421, in handle_reset
[rank0]:     self.env.reset(done_indices)
[rank0]:   File "/home/tony/workspace/ProtoMotions/phys_anim/envs/humanoid/common.py", line 506, in reset
[rank0]:     self.reset_envs(env_ids)
[rank0]:   File "/home/tony/workspace/ProtoMotions/phys_anim/envs/masked_mimic/tasks/base_task/common.py", line 62, in reset_envs
[rank0]:     super().reset_envs(env_ids)
[rank0]:   File "/home/tony/workspace/ProtoMotions/phys_anim/envs/humanoid/isaacgym.py", line 1146, in reset_envs
[rank0]:     self.compute_observations(env_ids)
[rank0]:   File "/home/tony/workspace/ProtoMotions/phys_anim/envs/masked_mimic/tasks/story/common.py", line 261, in compute_observations
[rank0]:     objects_bounding_box = self.object_id_to_object_bounding_box[self.scene_ids]
[rank0]: TypeError: 'method' object is not subscriptable
`

tesslerc, 2025-03-23:
Assumed this is solved, due to inactivity. Please re-open if you have any further issues.

closed 2025-03-23 by tesslerc