NVlabs/ProtoMotions #45: MaskMimic Error(config.yaml + AttributeError: 'MaskedMimicHumanoid' object has no attribute 'build_target_poses')

WesleyXu224, 2024-12-31:
Great Work！But i meet some errors when using maskmimic,i dont know how to solve this problem

First stage：PYTHON_PATH phys_anim/train_agent.py +exp=full_body_tracker +robot=smpl +backbone=isaacgym num_envs=128 motion_file=/home/group16/xuws/ProtoMotions/data/motions/smpl_humanoid_walk.npy
successfully get file   last.ckpt at /home/group16/xuws/ProtoMotions/results/full_body_tracker/last.ckpt

Second stage：PYTHON_PATH phys_anim/train_agent.py +exp=masked_mimic +robot=smpl +backbone=isaacgym num_envs=128 motion_file=/home/group16/xuws/ProtoMotions/data/motions/smpl_humanoid_walk.npy gt_actor_path=results/full_body_tracker

1、Error not find config.yaml：
Spawning 613c706879ee39c9132c018af3949fe.urdf ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 100% 0:00:00
{'disable_discriminator': True, 'discriminator_obs_historical_steps': 10, 'discriminator_obs_size_per_step': 232}
/home/group16/xuws/ProtoMotions/phys_anim/utils/motion_lib.py:150: UserWarning: To copy construct from a tensor, it is recommended to use sourceTensor.clone().detach() or sourceTensor.clone().detach().requires_grad_(True), rather than torch.tensor(sourceTensor).
  torch.tensor(key_body_ids, dtype=torch.long, device=device),
Loading motions from yaml/npy file
Loading 1/1 motion files: /home/group16/xuws/ProtoMotions/data/motions/smpl_humanoid_walk.npy
Loaded 1 motions with a total length of 5.667s.
Loaded 1 sub motions with a total trainable length of 5.667s.
Loading pre-trained full-body tracker from: results/full_body_tracker
/home/group16/xuws/ProtoMotions/phys_anim/agents/mimic_vae_dagger.py:59: FutureWarning: You are using `torch.load` with `weights_only=False` (the current default value), which uses the default pickle module implicitly. It is possible to construct malicious pickle data which will execute arbitrary code during unpickling (See https://github.com/pytorch/pytorch/blob/main/SECURITY.md#untrusted-models for more details). In a future release, the default value for `weights_only` will be flipped to `True`. This limits the functions that could be executed during unpickling. Arbitrary objects will no longer be allowed to be loaded via this mode unless they are explicitly allowlisted by the user via `torch.serialization.add_safe_globals`. We recommend you start setting `weights_only=True` for any use case where you don't have full control of the loaded file. Please open an issue on GitHub for any issues related to this experimental feature.
  pre_trained_expert = torch.load(checkpoint_path)
Error executing job with overrides: ['+exp=masked_mimic', '+robot=smpl', '+backbone=isaacgym', 'num_envs=128', 'motion_file=/home/group16/xuws/ProtoMotions/data/motions/smpl_humanoid_walk.npy', 'gt_actor_path=results/full_body_tracker']
[rank0]: Traceback (most recent call last):
[rank0]:   File "/home/group16/.conda/envs/mackmimic/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 92, in _call_target
[rank0]:     return _target_(*args, **kwargs)
[rank0]:   File "/home/group16/xuws/ProtoMotions/phys_anim/agents/mimic_vae_dagger.py", line 61, in __init__
[rank0]:     self.gt_actor_config = OmegaConf.load(
[rank0]:   File "/home/group16/.conda/envs/mackmimic/lib/python3.8/site-packages/omegaconf/omegaconf.py", line 189, in load
[rank0]:     with io.open(os.path.abspath(file_), "r", encoding="utf-8") as f:
[rank0]: FileNotFoundError: [Errno 2] No such file or directory: '/home/group16/xuws/ProtoMotions/results/full_body_tracker/lightning_logs/version_1/config.yaml'

[rank0]: The above exception was the direct cause of the following exception:

[rank0]: Traceback (most recent call last):
[rank0]:   File "phys_anim/train_agent.py", line 153, in <module>
[rank0]:     main()
[rank0]:   File "/home/group16/.conda/envs/mackmimic/lib/python3.8/site-packages/hydra/main.py", line 94, in decorated_main
[rank0]:     _run_hydra(
[rank0]:   File "/home/group16/.conda/envs/mackmimic/lib/python3.8/site-packages/hydra/_internal/utils.py", line 394, in _run_hydra
[rank0]:     _run_app(
[rank0]:   File "/home/group16/.conda/envs/mackmimic/lib/python3.8/site-packages/hydra/_internal/utils.py", line 457, in _run_app
[rank0]:     run_and_report(
[rank0]:   File "/home/group16/.conda/envs/mackmimic/lib/python3.8/site-packages/hydra/_internal/utils.py", line 223, in run_and_report
[rank0]:     raise ex
[rank0]:   File "/home/group16/.conda/envs/mackmimic/lib/python3.8/site-packages/hydra/_internal/utils.py", line 220, in run_and_report
[rank0]:     return func()
[rank0]:   File "/home/group16/.conda/envs/mackmimic/lib/python3.8/site-packages/hydra/_internal/utils.py", line 458, in <lambda>
[rank0]:     lambda: hydra.run(
[rank0]:   File "/home/group16/.conda/envs/mackmimic/lib/python3.8/site-packages/hydra/_internal/hydra.py", line 132, in run
[rank0]:     _ = ret.return_value
[rank0]:   File "/home/group16/.conda/envs/mackmimic/lib/python3.8/site-packages/hydra/core/utils.py", line 260, in return_value
[rank0]:     raise self._return_value
[rank0]:   File "/home/group16/.conda/envs/mackmimic/lib/python3.8/site-packages/hydra/core/utils.py", line 186, in run_job
[rank0]:     ret.return_value = task_function(task_cfg)
[rank0]:   File "phys_anim/train_agent.py", line 123, in main
[rank0]:     algo: PPO = instantiate(config.algo, env=env, fabric=fabric)
[rank0]:   File "/home/group16/.conda/envs/mackmimic/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 226, in instantiate
[rank0]:     return instantiate_node(
[rank0]:   File "/home/group16/.conda/envs/mackmimic/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 347, in instantiate_node
[rank0]:     return _call_target(_target_, partial, args, kwargs, full_key)
[rank0]:   File "/home/group16/.conda/envs/mackmimic/lib/python3.8/site-packages/hydra/_internal/instantiate/_instantiate2.py", line 97, in _call_target
[rank0]:     raise InstantiationException(msg) from e
[rank0]: hydra.errors.InstantiationException: Error in call to target 'phys_anim.agents.mimic_vae_dagger.MimicVAEDagger':
[rank0]: FileNotFoundError(2, 'No such file or directory')
[rank0]: full_key: algo

2、I find config.yaml under results/full_body_tracker,so i put it under version1，then Error occurs
'MaskedMimicHumanoid' object has no attribute 'build_target_poses'
 warnings.warn(f"enable_nested_tensor is True, but self.use_nested_tensor is False because {why_not_sparsity_fast_path}")
Epoch 0, collecting data... ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:--
Error executing job with overrides: ['+exp=masked_mimic', '+robot=smpl', '+backbone=isaacgym', 'num_envs=128', 'motion_file=/home/group16/xuws/ProtoMotions/data/motions/smpl_humanoid_walk.npy', 'gt_actor_path=results/full_body_tracker']
[rank0]: Traceback (most recent call last):
[... 8 lines cut ...]
[rank0]:   File "/home/group16/.conda/envs/mackmimic/lib/python3.8/site-packages/hydra/_internal/utils.py", line 223, in run_and_report
[rank0]:     raise ex
[rank0]:   File "/home/group16/.conda/envs/mackmimic/lib/python3.8/site-packages/hydra/_internal/utils.py", line 220, in run_and_report
[rank0]:     return func()
[rank0]:   File "/home/group16/.conda/envs/mackmimic/lib/python3.8/site-packages/hydra/_internal/utils.py", line 458, in <lambda>
[rank0]:     lambda: hydra.run(
[rank0]:   File "/home/group16/.conda/envs/mackmimic/lib/python3.8/site-packages/hydra/_internal/hydra.py", line 132, in run
[rank0]:     _ = ret.return_value
[rank0]:   File "/home/group16/.conda/envs/mackmimic/lib/python3.8/site-packages/hydra/core/utils.py", line 260, in return_value
[rank0]:     raise self._return_value
[rank0]:   File "/home/group16/.conda/envs/mackmimic/lib/python3.8/site-packages/hydra/core/utils.py", line 186, in run_job
[rank0]:     ret.return_value = task_function(task_cfg)
[rank0]:   File "phys_anim/train_agent.py", line 149, in main
[rank0]:     algo.fit()
[rank0]:   File "/home/group16/xuws/ProtoMotions/phys_anim/agents/ppo.py", line 265, in fit
[rank0]:     self.play_steps()
[rank0]:   File "/home/group16/xuws/ProtoMotions/phys_anim/agents/mimic.py", line 50, in play_steps
[rank0]:     super().play_steps()
[rank0]:   File "/home/group16/.conda/envs/mackmimic/lib/python3.8/site-packages/torch/utils/_contextlib.py", line 116, in decorate_context
[rank0]:     return func(*args, **kwargs)
[rank0]:   File "/home/group16/xuws/ProtoMotions/phys_anim/agents/ppo.py", line 344, in play_steps
[rank0]:     actor_outs = self.pre_env_step(obs, step)
[rank0]:   File "/home/group16/xuws/ProtoMotions/phys_anim/agents/mimic_vae_dagger.py", line 89, in pre_env_step
[rank0]:     gt_actor_inputs["mimic_target_poses"] = self.env.build_target_poses(
[rank0]: AttributeError: 'MaskedMimicHumanoid' object has no attribute 'build_target_poses'

tesslerc, 2025-01-01:
Thanks. Please pull the latest commit. `build_target_poses` was moved to `mimic_obs` "callback" function. The call in `mimic_vae_dagger` has been updated to reflect the new path.

tesslerc, 2025-03-23:
Assumed this is solved, due to inactivity. Please re-open if you have any further issues.

closed 2025-03-23 by tesslerc