NVlabs/ProtoMotions #46: if I want to adapt to a new type of humanoid robot use protomotion,what should we do

dbdxnuliba, 2025-01-01:
.Hello, if I want to adapt to a new type of humanoid robot use protomotion, what should I do? and what are the main steps involved，could you please help me list the steps,many thanks!

tesslerc, 2025-01-06:
1. Create config file: https://github.com/NVlabs/ProtoMotions/blob/main/phys_anim/config/robot/smpl.yaml
2. Obtain data in humanoid format. You can follow what we do in the retargeting script. You want the dof ordering to match the order that IsaacGym loads them.

The parameters for the first step can be obtained by loading the MJCF/URDF in IsaacGym and looking at:

num_bodies = self.gym.get_asset_rigid_body_count(humanoid_asset)
body_names = self.gym.get_asset_rigid_body_names(humanoid_asset)

num_dof = self.gym.get_asset_dof_count(humanoid_asset)
dof_names = self.gym.get_asset_dof_names(humanoid_asset)

num_joints = self.gym.get_asset_joint_count(humanoid_asset)

dbdxnuliba, 2025-01-07:
> 1. Create config file: https://github.com/NVlabs/ProtoMotions/blob/main/phys_anim/config/robot/smpl.yaml
> 2. Obtain data in humanoid format. You can follow what we do in the retargeting script. You [...]
many thanks,
1. create a config file like [smpl.yaml](https://github.com/NVlabs/ProtoMotions/blob/main/phys_anim/config/robot/smpl.yaml)
2. retarget the  self define robot following the  retargettting script **, could you please point out the  retargetting scripts** ,you mean we need this two steps

tesslerc, 2025-01-20:
Sorry for the delayed response.
You can see an example here: https://github.com/NVlabs/ProtoMotions/blob/main/data/scripts/convert_h1_to_isaac.py
This is how we retarget the AMASS dataset to the H1. It is adapted from LeCAR's OmniH2O effort.

Keep in mind that this is an example, and it is not optimal in any way.

closed 2025-03-23 by tesslerc