NVlabs/ProtoMotions #47: Isaacgym_body_names item not found when I training h1 in stage 2.

ImFroooost, 2025-01-09:
I trained h1 in stage 1, everything is OK; Then in stage 2, my input is:
 "python phys_anim/train_agent.py +exp=masked_mimic +robot=h1 +backbone=isaacgym motion_file=data/motions/h1_punch.npy gt_actor_path=results/full_body_tracker", 
 
 but it said :
"ConfigValueError('Item not found in ListConfig\n    full_key: config.robot.isaacgym_body_names\n    object_type=list')",

I checked data/scripts/retargeting/config.py, where i found the asset_file in h1_mapping() is "h1_no_head_no_hands.xml", so why the file_path doesn't match the robot_type? While the error still exists although i corrected the path, could you please explain the reason of this error? Thank you!

tesslerc, 2025-01-12:
There's a h1_maskedmimic config: https://github.com/NVlabs/ProtoMotions/blob/main/phys_anim/config/exp/h1_masked_mimic.yaml
Masked mimic uses a subset of bodies, so those are defined differently for each robot.

The current retargeting script expects an mjcf with the base robot structure, any added non articulated joints should be added externally via the "extension" data structure.

tesslerc, 2025-03-23:
Assumed this is solved, due to inactivity. Please re-open if you have any further issues.

closed 2025-03-23 by tesslerc