NVlabs/ProtoMotions #48: could you please provide sim2sim code when deploy the h1 policy

dbdxnuliba, 2025-01-13:
could you please provide sim2sim code when deploy the h1 policy

tesslerc, 2025-01-14:
We don't have any plans to integrate MuJoCo. However if you're interested in adding support, we'd happily accept a PR.

dbdxnuliba, 2025-01-24:
thanks for your replay ,can we train a policy for h1 only with vr data input  which only include left wrist , right wrist  and head pose ,if indeed ,then what's the corresponding command

tesslerc, 2025-01-24:
That's not supported "out of the box" but it should be fairly simple to adapt.
If you look at the mimic reward, it takes the full target pose and compares it with the full current pose. For rotations and translations.

You can adapt that to only consider the 3 joints you're interested in by selecting the 3 corresponding indices before computing the [mean-over-joints](https://github.com/NVlabs/ProtoMotions/blob/7e19fdc48a34e3196800bc47eb86e2dc3a37cc72/phys_anim/envs/mimic/mimic_utils.py#L72).

I'd suggest to mix in some full-body data through AMP. The discriminator will help fine-tune the overall posture to be more stable and realistic while the 3-point-tracking will help ensure it matches the target coordiates.

dbdxnuliba, 2025-01-25:
Thank you for your reply, but in the README, you mentioned   "Unitree H1 humanoid with end-effector and head joints made visible" 
 is it meanning when training the h1 , the motion file for the Unitree H1's .npy file only contains the poses of the head and end effector. Is it correct that it does not require the pose information of the whole body joints and the floating base as input?

tesslerc, 2025-01-26:
1. The motion file contains the coordinates and rotations of all joints. In this modification you also have the head, toes, and wrist positions. Since they aren't actuated, in the original config you had to infer the wrist position based on the elbow position + orientation.
2. The numpy is a dictionary. Look at how we load it and parse it in the h1 motion lib file.
3. https://github.com/NVlabs/ProtoMotions/blob/main/data/scripts/convert_h1_to_isaac.py this is an example of how we converted for our demonstrations.