NVlabs/ProtoMotions #50: possibility to record and export the data?

leob03, 2025-01-17:
I noticed the record_motion config option, how do I record the generated motion of an agent inside the Isaac env? Is there a possibility/script to export that recorded motion into a .bvh format?

tesslerc, 2025-01-20:
This code is provided as an example. We have used it in the past, but it may have lost alignment with recent codebase upgrades.
If you test and find any issues, let us know and we'll fix.

At a high level. This records multiple environment in parallel. For each environment it will record the sequence of poses (trajectory).
You can then pick to store in poselib format or in the AMASS format. Using the flag `store_poselib`.
If there are objects in the scene, there's an example of how we also store the object information to enable reconstruction of the full scene. For the terrain, it also has a function for saving the heightmap.

We don't have a script to convert to bvh/fbx, but if you extend the recording to support that functionality we'd be very happy to accept a PR.

leob03, 2025-01-20:
Got it, thank you for your quick answer! I'm focusing on another project currently but might try to work on the extension in a close future.

closed 2025-01-26 by tesslerc