NVlabs/ProtoMotions #51: err train masked_mimic for h1   in stage 2

dbdxnuliba, 2025-01-26:
when run command to train h1 with masked_mimic in stage 2,it occurs err:

**Error executing job with overrides: ['+exp=masked_mimic', '+robot=h1', '+backbone=isaacgym', 'motion_file=phys_anim/data/motions/h1_punch.npy', 'num_envs=1024', 'algo.config.batch_size=1024', 'force_flat_terrain=True', '+opt=wandb', 'gt_actor_path=results/full_body_tracker']
Error in call to target 'phys_anim.envs.masked_mimic.isaacgym.MaskedMimicHumanoid':
ConfigValueError('Item not found in ListConfig\n    full_key: config.robot.isaacgym_body_names\n    object_type=list')
full_key: env
Item not found in ListConfig
    full_key: config.robot.isaacgym_body_names
    object_type=list

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.**

Here is the detail:

```

(ProtoMotions) rob@rob:~/ProtoMotions$ PYTHON_PATH phys_anim/train_agent.py +exp=masked_mimic +robot=h1 +backbone=isaacgym motion_file=phys_anim/data/motions/h1_punch.npy  num_envs=1024 algo.config.batch_size=1024 force_flat_terrain=True +opt=wandb gt_actor_path=results/full_body_tracker
Importing module 'gym_38' (/home/rob/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/rob/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
wandb: WARNING This integration is tested and supported for lightning Fabric 2.1.3.
wandb: WARNING             Please report any issues to https://github.com/wandb/wandb/issues with the tag `lightning-fabric`.
2025-01-26 14:22:11,191 - INFO - logger - logger initialized
phys_anim/train_agent.py:67: UserWarning: 
The version_base parameter is not specified.
Please specify a compatability version level, or None.
Will assume defaults for version 1.1
  @hydra.main(config_path="config", config_name="base")
/home/rob/anaconda3/envs/ProtoMotions/lib/python3.8/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'base': Defaults list is missing `_self_`. See https://hydra.cc/docs/1.2/upgrades/1.0_to_1.1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
/home/rob/anaconda3/envs/ProtoMotions/lib/python3.8/site-packages/hydra/_internal/hydra.py:119: UserWarning: Future Hydra versions will no longer change working directory at job runtime by default.
See https://hydra.cc/docs/1.2/upgrades/1.1_to_1.2/changes_to_job_working_dir/ for more information.
  ret = run_job(
Initializing distributed: GLOBAL_RANK: 0, MEMBER: 1/1
----------------------------------------------------------------------------------------------------
distributed_backend=nccl
All distributed processes registered. Starting with 1 processes
----------------------------------------------------------------------------------------------------

[rank: 0] Seed set to 0
Setting seed: 0
PyTorch version 2.2.0
Device count 3
/home/rob/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/rob/.cache/torch_extensions/py38_cu121 as PyTorch extensions root...
Emitting ninja build file /home/rob/.cache/torch_extensions/py38_cu121/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
Will spawn a total of 339 objects.
Will spawn 339 scenes.
/home/rob/anaconda3/envs/ProtoMotions/lib/python3.8/site-packages/torch/functional.py:507: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at /opt/conda/conda-bld/pytorch_1704987301168/work/aten/src/ATen/native/TensorShape.cpp:3549.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: enabled
Creating ground plane
Ground plane spawned
Warning: participates_in_heightmap is not a valid option for object asset
Loading object assets... ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:--Using VHACD cache directory '/home/rob/.isaacgym/vhacd'
Started convex decomposition for mesh 'data/assets/scenes/train/SAMP/LargeSofas/a8db9db73bb5a4dbf51f77a6d7299806.obj'
Finished convex decomposition for mesh 'data/assets/scenes/train/SAMP/LargeSofas/a8db9db73bb5a4dbf51f77a6d7299806.obj': 1 hulls

[2025-01-26 14:38:16,704][trimesh][WARNING] - only got 117/256 samples!
[2025-01-26 14:38:16,715][trimesh][WARNING] - only got 57/256 samples!
Spawning 613c706879ee39c9132c018af3949fe.urdf ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 100% 0:00:00
{'disable_discriminator': True, 'discriminator_obs_historical_steps': 10, 'discriminator_obs_size_per_step': 158}
/home/rob/ProtoMotions/phys_anim/utils/motion_lib.py:150: UserWarning: To copy construct from a tensor, it is recommended to use sourceTensor.clone().detach() or sourceTensor.clone().detach().requires_grad_(True), rather than torch.tensor(sourceTensor).
  torch.tensor(key_body_ids, dtype=torch.long, device=device),
Loading motions from yaml/npy file
Loading 1/1 motion files: phys_anim/data/motions/h1_punch.npy
Loaded 1 motions with a total length of 6.300s.
Loaded 1 sub motions with a total trainable length of 6.300s.
Error executing job with overrides: ['+exp=masked_mimic', '+robot=h1', '+backbone=isaacgym', 'motion_file=phys_anim/data/motions/h1_punch.npy', 'num_envs=1024', 'algo.config.batch_size=1024', 'force_flat_terrain=True', '+opt=wandb', 'gt_actor_path=results/full_body_tracker']
Error in call to target 'phys_anim.envs.masked_mimic.isaacgym.MaskedMimicHumanoid':
ConfigValueError('Item not found in ListConfig\n    full_key: config.robot.isaacgym_body_names\n    object_type=list')
full_key: env
Item not found in ListConfig
    full_key: config.robot.isaacgym_body_names
    object_type=list

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
```

tesslerc, 2025-01-26:
Not sure if this is your issue, but for the H1 you should use the `exp=h1_masked_mimic` which loads this experiment file https://github.com/NVlabs/ProtoMotions/blob/main/phys_anim/config/exp/h1_masked_mimic.yaml

dbdxnuliba, 2025-02-06:
thanks, but re run meet another issue 

(ProtoMotions) rob@rob:~/ProtoMotions$ PYTHON_PATH phys_anim/train_agent.py +exp=h1_masked_mimic +robot=h1 +backbone=isaacgym motion_file=phys_anim/data/motions/h1_punch.npy  num_envs=1024 algo.config.batch_size=1024 force_flat_terrain=True +opt=wandb gt_actor_path=results/full_body_tracker
Importing module 'gym_38' (/home/rob/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/rob/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
wandb: WARNING This integration is tested and supported for lightning Fabric 2.1.3.
wandb: WARNING             Please report any issues to https://github.com/wandb/wandb/issues with the tag `lightning-fabric`.
2025-02-06 09:53:35,763 - INFO - logger - logger initialized
phys_anim/train_agent.py:67: UserWarning: 
The version_base parameter is not specified.
Please specify a compatability version level, or None.
Will assume defaults for version 1.1
  @hydra.main(config_path="config", config_name="base")
/home/rob/anaconda3/envs/ProtoMotions/lib/python3.8/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'base': Defaults list is missing `_self_`. See https://hydra.cc/docs/1.2/upgrades/1.0_to_1.1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
In 'exp/h1_masked_mimic': Could not load 'opt/legged_robot'.

Config search path:
	provider=hydra, path=pkg://hydra.conf
	provider=main, path=file:///home/rob/ProtoMotions/phys_anim/config
	provider=schema, path=structured://

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.

tesslerc, 2025-02-06:
Removing the `legged_robot` opt is the right thing. I must have missed that in one of the code updates.
Can you pull the latest code from git? The error you are now seeing refers to an old version of the code -- see [here](https://github.com/NVlabs/ProtoMotions/blob/7e19fdc48a34e3196800bc47eb86e2dc3a37cc72/phys_anim/agents/mimic_vae_dagger.py#L88)

dbdxnuliba, 2025-02-06:
many thanks, the command run well now ,      by

            # gt_actor_inputs["mimic_target_poses"] = self.env.build_target_poses(
            #     num_future_steps=self.gt_actor_config.env.config.mimic_target_pose.num_future_steps,
            #     target_pose_type=self.gt_actor_config.env.config.mimic_target_pose.type,
            #     with_time=self.gt_actor_config.env.config.mimic_target_pose.with_time,
            #     env_ids=torch.arange(self.env.num_envs, device=self.device),
            # )
            gt_actor_inputs["mimic_target_poses"] = (
                self.env.mimic_obs_cb.build_target_poses(
                    num_future_steps=self.gt_actor_config.env.config.mimic_target_pose.num_future_steps,
                    target_pose_type=self.gt_actor_config.env.config.mimic_target_pose.type,
                    with_time=self.gt_actor_config.env.config.mimic_target_pose.with_time,
                    env_ids=torch.arange(self.env.num_envs, device=self.device),
                )
            )

> Removing the `legged_robot` opt is the right thing. I must have missed that in one of the code updates. Can you pull the latest code from git? The error you are now seeing refers to an old version [...]

closed 2025-02-06 by dbdxnuliba