NVlabs/ProtoMotions #52: Addition of root_pos_offset twice in retargeting

RaiAnant, 2025-01-30:
Hi,

I was trying to understand the logic behind the retargeting code in `fit_smpl_motion.py`. I was confused as to why are we adding `root_pos_offset` twice after the otptimisation loop when passing root_translation to fk_batch.

First we compute `root_trans_offset_dump` [here] (https://github.com/NVlabs/ProtoMotions/blob/7e19fdc48a34e3196800bc47eb86e2dc3a37cc72/data/scripts/retargeting/fit_smpl_motion.py#L188)  by adding `root_pos_offset` to `root_trans_offset`.

But then later when running forward kinematics, we again add `root_pos_offset` to `root_trans_offset_dump` [here](https://github.com/NVlabs/ProtoMotions/blob/7e19fdc48a34e3196800bc47eb86e2dc3a37cc72/data/scripts/retargeting/fit_smpl_motion.py#L197C1-L201C6)

Would really love some clarification to this. Thanks!

tesslerc, 2025-02-06:
Indeed this looks like a mistake.
If you remove the second one does it lead to any issues?

RaiAnant, 2025-02-08:
I tried to remove it and test it on a couple of examples but it didnt make a lot of difference, probably because in those cases `root_pos_offset` was relatively very small compared to `root_trans_offset`. Still wanted to double check the logic.

tesslerc, 2025-02-11:
Thanks
We're currently using Mink retargeter, which seems to be working better. And the whole process is a bit cleaner (I think).
Retargeting pipeline will be released alongside v2 in the near future.

closed 2025-02-21 by RaiAnant