NVlabs/ProtoMotions #54: how can we export actor policy and depoly it in the real world for h1  MaskedMimic

dbdxnuliba, 2025-02-06:
after train  h1  MaskedMimic 
how can we export actor policy and depoly it in the real world  or geneis simulator for h1  MaskedMimic

tesslerc, 2025-02-06:
We don't have an example for deploying, so I can't help with that.
If you have a real robot and are interested in deploying, I'd be happy to chat (please contact me via email) and try figure a path together.

Regarding Genesis, I believe we'll add support. But can't commit to any timeline.

Hellod035, 2025-02-15:
The correct approach is that during distillation, all observations must be available in the real world, which is a necessary condition for sim2real.

tesslerc, 2025-03-23:
Assumed this is solved, due to inactivity. Please re-open if you have any further issues.

closed 2025-03-23 by tesslerc

minolta001, 2025-04-07:
Hi @tesslerc @dbdxnuliba, I have a Unitree H1-2 robot and I am also working on policy deployment to real robot. Do you mind having a discussion or sharing some insights?

tesslerc, 2025-04-07:
I haven't deployed on real yet, due to lack of physical robot. But would be very happy to help in which ever way I can.

minolta001, 2025-04-08:
@tesslerc Thanks! May I send you an email later (probably within this week) for introducing the situation on my side? 
@dbdxnuliba Please feel free to discuss if you would like to, or please let me know if you want to chat on email :)

tesslerc, 2025-04-09:
Yes, that would be fine.