NVlabs/ProtoMotions #56: Hydra config error in training

canoneod, 2025-02-19:
Hello, I'm trying out newly pushed v2 over original repository, and have encountered an issue when I'm trying to train stage 1.

In config/env/base_env.yaml,

```
# @package _global_

defaults:
  - /terrain/flat
  - /motion_manager/base_manager

# Env structure
env:
  _target_: protomotions.envs.base_env.env.BaseEnv
  _recursive_: False
  config:
    experiment_name: ${experiment_name}
    num_envs: ${num_envs}
    headless: ${headless}
    
    simulator: ${simulator}
```
So I assumed keys in simulatr/base.yaml will show up on the run. But when instantiating env (`env = instantiate(config.env, device=fabric.device)`    in `train_agent.py`) when I check below part,

```
        SimulatorConfigClass = get_class(self.config.simulator._config_target_)
        simulator_config: SimulatorConfig = SimulatorConfigClass.from_dict(self.config.simulator.config)
        SimulatorClass = get_class(self.config.simulator._target_)    
```
There is no _config_target_ nor _target _ keys in `self.config.simulator`   (which is `self.config.env` in train_agent.py) and I'm getting error 

```
hydra.errors.InstantiationException: Error in call to target 'protomotions.envs.mimic.env.Mimic':
ConfigAttributeError('Missing key _config_target_\n    full_key: config.simulator._config_target_\n    object_type=dict')
full_key: env
```

When I print `config.env.config.simulator`, in train_agent.py it only has below parameters, neither `_config_target_` nor `_target_`

`{'sim': {'physx': {'num_position_iterations': 8, 'num_velocity_iterations': 1, 'default_buffer_size_multiplier': 10, 'contact_offset': 0.005, 'rest_offset': 0.0001}}}`

I added arguments correctly for training, and what could be the issue?

Also, I've recognized pretrained model of stage 2 disappeared on latest commit. Is it intended? 

Thanks in advance.

tesslerc, 2025-02-19:
Can you share the full command?

Can you check this command from the protomotions directory?
`python protomotions/train_agent.py +exp=full_body_tracker/transformer_flat_terrain +robot=smpl +simulator=isaacgym motion_file=data/motions/smpl_humanoid_walk.npy +experiment_name=full_body_tracker_test`

Regarding the model, that's indeed the case. We are currently in advanced stages of training new models that match the new codebase. Once those are ready we will release them in the repo.

canoneod, 2025-02-20:
The provided command works just fine. It seems like the problem occurs when changed to `+robot=smplx` and motion file accordingly.

Same problem occurred with below command

```
python protomotions/train_agent.py +exp=full_body_tracker/transformer_flat_terrain +robot=smplx +simulator=isaacgym motion_file=data/motions/smplx_humanoid_walk.npy +experiment_name=full_body_tracker_test
```

tesslerc, 2025-02-20:
Thanks! We must have missed that during the code migration.
You can safely remove it. It's only important if you want to work on human-object interactions. In that case it might help prevent physx artifacts.

canoneod, 2025-02-21:
Thanks! Helped a lot.

closed 2025-02-21 by canoneod