NVlabs/ProtoMotions #58: Some questions regarding the retargeting pipeline

CavalloneChen, 2025-02-22:
Congratulations on the V2 update! I have some questions regarding the motion retargeting (from AMASS to Unitree G1) pipeline:

1. Is the coordinate system of the retargted motions different from that of the PHC retargeted motions? I tried to run PHC training on the motions retargeted from ProtoMotions (V1 pipeline to be specific), but the result seems to be very weired.
2. Is there any instructions on how to utilize the brand new Mink pipline to retarget from AMASS to Unitree H1/G1 robots? Will the output contain the same set of keys as ProtoMotions V1 pipeline? And will the coordinate system be the same as V1 ?

Thank you for your time.

CavalloneChen, 2025-02-23:
Thank you for the great news. I noticed that CMU's LeCAR-Lab has recently unveiled two interesting repos https://github.com/LeCAR-Lab/HumanoidVerse and https://github.com/LeCAR-Lab/ASAP . Maybe you can find some G1 related resources there.

Saluze, 2025-04-02:
> Thank you for the great news. I noticed that CMU's LeCAR-Lab has recently unveiled two interesting repos https://github.com/LeCAR-Lab/HumanoidVerse and https://github.com/LeCAR-Lab/ASAP . Maybe you can find some G1 related resources there.

Hi, have you ever tried  the motion retargeting from AMASS to Unitree G1? I‘m’ suffering from some problems that were difficult to solve for me during the attempt.

tesslerc, 2025-04-02:
> Hi, have you ever tried the motion retargeting from AMASS to Unitree G1? I‘m’ suffering from some problems that were difficult to solve for me during the attempt.

We just pushed v2.3.2 adding support for the Unitree G1.
There's also an example motion in the data/motions directly.
This could be helpful. We have successfully trained this robot to walk and cartwheel.

The retargeting pipeline is identical to the H1, you define the robot_type to be the g1 and it will utilize Mink for retargeting.

closed 2025-06-04 by tesslerc