NVlabs/ProtoMotions #59: How can I train full body tracker with scene interaction?

bowieshi, 2025-02-23:
Hi. I tested ProtoMotions V2. But I can not find where I can train my full body tracker with samp data. The config file does not indicate where is samp scene data, like following:
`# @package _global_

defaults:
  - /agent/mimic/agent
  - /env/mimic

  - /agent/ppo/models/transformer_actor
  - /agent/ppo/models/mlp_critic_large
  - /agent/mimic/models/transformer_actor_target_pose
  - /agent/mimic/models/mlp_critic_target_pose
  # - /scenes/samp
  - /agent/mimic/options/global_tracking
  - /agent/mimic/options/early_termination_tracking_err
  - /agent/mimic/options/dynamic_sampling

env:
  config:
    mimic_target_pose:
      type: max-coords-future-rel
      with_time: true
      num_future_steps: 15
`
The `scenes/samp` is commented. And I can not find samp yaml file in the codebase. This function is supported in V1 version. Can I inquire how can I include samp data as training data in V2 version?

tesslerc, 2025-02-24:
This is correct.
v1 had SAMP support, but was also hard-coded around how SAMP is constructed.
Where we are moving towards with v2 is a more general framework.
Now scenes aren't global entities anymore (as this seems to cause simulation issues in certain cases). Instead, they are assigned per-humanoid.

We intend to extended the code to fully support motion tracking within general scenes.

bowieshi, 2025-02-24:
> This is correct. v1 had SAMP support, but was also hard-coded around how SAMP is constructed. Where we are moving towards with v2 is a more general framework. Now scenes aren't global entities [...]

Thank you for your response! Can I inquire how can I import samp or other custom scene. Is there any guidance or instruction about training with scene? Also, during training with V1, I met error that the training crashed down because `CUDA error: an illegal memory access was encountered`. But I was using A100 with 80GB memory and used flat ground, num_envs=1024. I have trained for three times. But each time the training would crash after about 22 hours. Can I inquire do you know any possible issue about this training failure? Thank you for your help!

```
5539 Epoch 2760, collecting data... ━━╸━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   6% 0:00:23
5540 
5541 Error executing job with overrides: ['+exp=full_body_tracker', '+robot=smplx', '+backbone=isaacgym', 'motion_file=/media/xuzhao/Partition5/qyzheng/boao/AMASS-X/motion_file.pt', 'force_flat_terrain=True', 'algo.config.batch_size=4096', 'num_envs=1024', '+opt=wdb']
5542 Traceback (most recent call last):
5543   File "phys_anim/train_agent.py", line 123, in main
5544     algo.fit()
5545   File "/home/qyzheng/boao/ProtoMotions/phys_anim/agents/ppo.py", line 259, in fit
5546     self.play_steps()
5547   File "/home/qyzheng/boao/ProtoMotions/phys_anim/agents/mimic.py", line 50, in play_steps
5548     super().play_steps()
5549   File "/home/qyzheng/anaconda3/envs/isaac/lib/python3.8/site-packages/torch/utils/_contextlib.py", line 115, in decorate_context
5550     return func(*args, **kwargs)
5551   File "/home/qyzheng/boao/ProtoMotions/phys_anim/agents/ppo.py", line 343, in play_steps
5552     actor_state = self.env_step(actor_state)
5553   File "/home/qyzheng/boao/ProtoMotions/phys_anim/agents/ppo.py", line 429, in env_step
5554     obs, rewards, dones, extras = self.env.step(actor_state["actions"])
5555   File "/home/qyzheng/boao/ProtoMotions/phys_anim/envs/base_interface/isaacgym.py", line 157, in step
5556     self.post_physics_step()
5557   File "/home/qyzheng/boao/ProtoMotions/phys_anim/envs/mimic/common.py", line 697, in post_physics_step
5558     self.motion_times += self.dt
5559 RuntimeError: CUDA error: an illegal memory access was encountered
5560 CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
5561 For debugging consider passing CUDA_LAUNCH_BLOCKING=1.
5562 Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.
5563 
5564 
5565 Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
```

tesslerc, 2025-02-26:
@bowieshi , see [here](https://github.com/NVlabs/ProtoMotions/tree/main/examples/scene_mimic) we have added an example for extending the mimic env with a single object + motion.
This example is in IsaacLab. It will kinematically replay the motion of a humanoid walking and sitting on a chair.

tesslerc, 2025-03-23:
Assumed this is solved, due to inactivity. Please re-open if you have any further issues.