NVlabs/ProtoMotions #61: Func compute_observations bug

Huisouan, 2025-02-25:
When I tried to use func compute_humanoid_observations.It seems that 'GenesisSimulator' object has no attribute 'key_body_ids'.Which should be attribute of env.

[image]

Huisouan, 2025-02-25:
Plus,if  bucket_width <= 1 there will be CUDA error

[image]

tesslerc, 2025-02-26:
Thanks for the feedback.
I see the first issue indeed occurs when the observations are not set to max_coords_obs. We haven't used that configuration in a while.
You can fix by changing to `self.env.key_body_ids`. I'd expect some mismatch in the expected observation size that would need fixing in the config.

Regarding the prioritization buckets, we have removed this logic in the latest v2.1 update.
While this provides some more granularity in prioritization, it is very tricky to tune and may cause unwanted behavior in some edge cases.
We now use the periodic evaluations to set the motion priorities. This should be more stable and prevent the crash you experienced.

Huisouan, 2025-02-27:
Thank you for your response. I noticed that when the configurations of `obs` and `ref_obs` differ, the `history_buffer` encounters a tensor shape mismatch issue during reset. Perhaps the `history_buffer` could be initialized differently based on whether a dataset is used, with a separate buffer specifically designed for datasets. I am really looking forward to the 2.1 version update. Could you provide any information on the update schedule?

tesslerc, 2025-02-27:
v2.1 was released yesterday with the motion-prioritization fixes.

If I understand correctly, the issue you're encountering is during `reset_hist_ref` since [this line](https://github.com/NVlabs/ProtoMotions/blob/81167863d6c8129c1e50613da4e6a9259c02a29f/protomotions/envs/base_env/components/humanoid_obs.py#L107) is hard coded to use the max-observations. Correct?
Seems like `build_self_obs_demo` is also hard-coded to the same max-obs format.

The fix seems to be making it all configurable. Need to dig into the various dependencies to see which of these must have the same format and which can be independent.

Huisouan, 2025-03-03:
Thank you for your prompt response. 
I noticed that dynamic_sampling has been removed in version v2.1. This feature previously utilized bucket-based segmentation for action fragments. I would like to inquire whether removing this functionality could negatively impact the training of specific actions? 
Given the varying difficulty levels among actions in the dataset, would failure to employ dynamic sampling - which increases sampling probability for challenging actions - result in unsuccessful training outcomes for such actions? For instance, actions requiring precise jumping maneuvers might struggle to achieve successful execution without appropriate data slicing mechanisms.

tesslerc, 2025-03-03:
We didn't entirely remove it, but rather over-sample entire motion clips instead of portions within them.
In certain edge cases, the bucketing method can actually harm performance.

One such example is due to data imperfections. The bucketing technique can lead to dramatic over-sampling.
We changed it to the more standard evaluate - then over-sample.
We evaluate the policy every 200 epochs and over-sample entire motions that have failed.

Huisouan, 2025-03-04:
Thank you very much for your reply. Wish you a wonderful day!

tesslerc, 2025-03-23:
Assumed this is solved, due to inactivity. Please re-open if you have any further issues.

closed 2025-03-23 by tesslerc