NVlabs/ProtoMotions #65: Some Questions and Discussions on the Capabilities of the MaskedMimic Model

Ccarrotr, 2025-03-07:
Hello, I sincerely appreciate your outstanding work—it has provided me with great inspiration and valuable insights. I have some questions and thoughts that I would love to discuss with you. My understanding may be shallow or even naive, and I hope you won’t mind.

First, since I have updated to version 2.0, I am eager to access the pre-trained model as soon as possible. Due to my hardware limitations and computational constraints, training the model requires an extensive amount of time. However, I am very eager to quickly test its capabilities.

Second, I would like to discuss the model’s ability to construct in-between motion sequences. I have read many of your previous works, including the seminal DeepMimic paper, where goal-reaching tasks are constrained by reward functions. My question is: when adopting a generative model approach, how well does the model perform in long-distance goal tracking?

In generative models, it seems challenging to incorporate a controlled approach, making it difficult to constrain the agent to perform the desired actions, especially when the target state is far from the agent's current state. In your paper, it appears that the model can generate certain in-between motions through the temporal masking mechanism. However, if the target is not present in the near-future reference motions, can the agent still move toward the desired position and body posture?

Additionally, my current research goal is to generate an intermediate motion sequence given an initial and final full-body pose, ensuring that the final motion reliably lands at the desired target state. I aim to achieve this by integrating reinforcement learning with generative modeling. Given that MaskedMimic's training is divided into two stages—the first being reinforcement learning and the second appearing to follow a purely generative approach—does this separation lead to a loss of motion generation capabilities that reinforcement learning would otherwise provide?

These are my main questions and thoughts. I sincerely look forward to your response.

tesslerc, 2025-03-09:
Thanks for the kind words.

We have just open sourced the motion tracker. We will also open source the MaskedMimic model compatible with v2, very soon.

Regarding inbetweening. Yes, it is possible. The further away (in space and time) the less precise it tends to be. But from what we've seen, the model does reaching very close to the provided target pose.

Maybe using RL to fine-tune the model, similar to RLFT/HF schemes in LLMs, could be useful here to make the model reach the target pose with higher precision.

tesslerc, 2025-04-11:
Closed due to inactivity.
Feel free to re-open if still relevant.