NVlabs/ProtoMotions #67: Using text description as input

canoneod, 2025-03-19:
Hello, upon the recent update on the maskedmimic pretrained model, I wonder it would support an evaluation with text description provided as in https://github.com/NVlabs/ProtoMotions/issues/35. It used to have a `user_control` module which I could input a text via json file.

Many thanks to wonderful work!

tesslerc, 2025-03-19:
We do plan to re-introduce ways for controlling the character. Text being one of them.
I'm not sure what would be the expected timeline for this. Our goal with any released components in v2 is to have them as general as possible, so they play nicely with other methods and can be extended.

canoneod, 2025-03-22:
Thanks for the answer!

closed 2025-03-22 by canoneod