NVlabs/ProtoMotions #69: error train smpl with MaskedMimic in stage 1

YunyangZhao, 2025-03-23:
I encountered an out-of-memory error during the first stage of training with MaskedMimic. Do I need to modify the code, or is it simply due to insufficient hardware and requires an upgrade? My computer has a GeForce 4090 GPU with 24GB of memory.Thank you！

[image]

[image]

tesslerc, 2025-03-23:
The default parameters will use 4096 humanoids and a batch size of 16384.
You can reduce those until they fit your GPU memory. I'd suggest to try keeping a ratio of 4:1 in terms of batch size versus humanoids.

YunyangZhao, 2025-03-24:
thank you very much!

closed 2025-03-24 by YunyangZhao