NVlabs/ProtoMotions #70: ASE

ghassen-fatnassi, 2025-03-23:
is the sword and shield humanoid not fully tested on isaac lab still? or should i be good to go?

tesslerc, 2025-03-23:
I believe it should work, the USD asset is there and it should load.
However, we haven't fully tested this combination yet.

If you encounter any issues, please let us know and we'll investigate.

ghassen-fatnassi, 2025-03-23:
thanks a lot! I'm working on adding self-play to see how far the CALM/ASE model goes , the codebase is scaring me haha , I'll try to manage i guess

closed 2025-03-23 by ghassen-fatnassi

tesslerc, 2025-03-24:
True, the code introduces new concepts, compared to the standard RLGames from CALM/ASE/AMP.
But we've found it to be much easier to build upon, and build fast, once you get to know the basic building blocks.

If you have any questions, I'd be happy to hop on a call, or answer here/email.
So feel free to reach out!

ghassen-fatnassi, 2025-03-24:
that's amazing , I'll definetly keep in touch , thanks a lot <3