NVlabs/ProtoMotions #71: IsaacLab simulator error

xiaotangzh, 2025-03-23:
Thanks for you excellent work! But I met an error when I run examples/isaaclab_spawning_scenes:

```
Traceback (most recent call last):
  File "D:\ProtoMotions\examples\isaaclab_spawning_scenes.py", line 197, in <module>
    simulator = IsaacLabSimulator(config=simulator_config, terrain=terrain, scene_lib=scene_lib, visualization_markers=None, device=device, simulation_app=simulation_app)
  File "d:\protomotions\protomotions\simulator\isaaclab\simulator.py", line 107, in __init__
    scene_cfg = self._get_scene_cfg()
  File "d:\protomotions\protomotions\simulator\isaaclab\simulator.py", line 135, in _get_scene_cfg
    scene_cfg = SceneCfg(
  File "d:\protomotions\protomotions\simulator\isaaclab\utils\scene.py", line 120, in __init__
    actuators = {
  File "d:\protomotions\protomotions\simulator\isaaclab\utils\scene.py", line 123, in <dictcomp>
    effort_limit=robot_config.dof_effort_limits[i],
TypeError: 'NoneType' object is not subscriptable
```

tesslerc, 2025-03-23:
Thanks for raising this issue.
Just pushed a fix. 
Please let me know if you encounter any additional issues!

xiaotangzh, 2025-03-24:
Thanks for your timely update. I met another issue with Hydra:
```
(protomotions) D:\ProtoMotions>D:\IsaacLab\isaaclab.bat -p protomotions/train_agent.py +exp=steering_mlp +robot=h1 +simulator=isaaclab +experiment_name=h1_steering
[INFO] Using python from: C:\Users\zxtst\anaconda3\envs\protomotions\python.exe
2025-03-24 13:35:29,322 - INFO - logger - logger initialized
INFO:poselib:logger initialized
D:\ProtoMotions\protomotions\train_agent.py:72: UserWarning:
The version_base parameter is not specified.
Please specify a compatability version level, or None.
Will assume defaults for version 1.1
  @hydra.main(config_path="config", config_name="base")
Error parsing override '+exp'
missing EQUAL at '<EOF>'
See https://hydra.cc/docs/1.2/advanced/override_grammar/basic for details

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
``` 

I notice that the Hrdra versions in requirements.txt are different for IsaacLab and IsaacSim, would this be the reason?

closed 2025-06-04 by tesslerc