NVlabs/ProtoMotions #73: make .npy file like smpl_walk_humanoid.npy

sunmengmm, 2025-03-25:
hi, thank u for your contribution!
i do record human motion in a video, and I collect some data from TRAM(Global Trajectory and Motion of 3D Humans from in-the-wild Videos), its outputs include data like:hps_track.npy(pred_pose,pred_rotmat,pred_trans...) and camera.npy(world_cam_R,world_cam_T....)

AND i want to transfer these outputs into a file like smpl_walk_humanoid.npy in /ProtoMotions/data/motions/.

Why the order of body_names in /ProtoMotions/protomotions/config/robot/smpl.yaml IS quite diffierent from the default order in the SMPL MODEL. 

In smpl.yaml, you set  body_names: ['Pelvis', 'L_Hip', 'L_Knee', 'L_Ankle', 'L_Toe', 'R_Hip', 'R_Knee', 'R_Ankle', 'R_Toe', 'Torso', 'Spine', 'Chest', 'Neck', 'Head', 'L_Thorax', 'L_Shoulder', 'L_Elbow', 'L_Wrist', 'L_Hand', 'R_Thorax', 'R_Shoulder', 'R_Elbow', 'R_Wrist', 'R_Hand']
BUT WE HAVE the default order in the SMPL MODEL:['Pelvis', 'L_Hip', 'R_Hip', 'Torso', 'L_Knee',  'R_Knee', 'Spine', 'L_Ankle', 'R_Ankle', 'Chest','L_Toe', 'R_Toe', 'Neck', 'L_Thorax', 'R_Thorax',  'Head', 'L_Shoulder', 'R_Shoulder', 'L_Elbow',  'R_Elbow', 'L_Wrist', 'R_Wrist', 'L_Hand', 'R_Hand']
Altough i did the permutations ,the play_motion.py shows my motion file may not  converted correctly.

Hope u can give me some advice in make motion files like smpl_walk_humanoid.npy. Thank you so much.

sunmengmm, 2025-03-26:
> See this section in the readme: https://github.com/NVlabs/ProtoMotions?tab=readme-ov-file#convert-the-motions-to-motionlib-format
> 
> You need to save your file in the AMASS format and body ordering. Then, you run `convert_amass_to_isaac` (you might need to slightly tweak it to load your specific file) which will re-order the joints and convert everything to our format. This also handles converting the humanoid to z-axis up.

Thank you for your advice , i do make some progress.
First i save my file in the AMASS format, and run covert_amass_to_isaac.py, then  i run play_motion.py, the robot appears to be floating in the air  and facing the ground, as shown in the following image:

[image] 
I tried correction = R.from_euler('x', -90, degrees=True) ,and the result changed to this

[image]
In both of these two situations, my robots seems to remain in the air.
Could you give me some advice? Or do you think the issue still lies in the conversion of my data into the AMASS format?

tesslerc, 2025-03-26:
The default AMASS format is Y-up. If your data is in Z-up already, then you can comment out [this](https://github.com/NVlabs/ProtoMotions/blob/14368ee81ed2dd64c8de90808c3b98dfb641d01b/data/scripts/convert_amass_to_isaac.py#L352) line and try.

When we load the motions we automatically fix the height such that the lowest joint in any frame will touch the ground. Are all frames indeed in the air like you show in this image? Could there be a "rogue" frame that somehow has a height mismatch?

XuanchaoPENG, 2025-03-27:
> > See this section in the readme: https://github.com/NVlabs/ProtoMotions?tab=readme-ov-file#convert-the-motions-to-motionlib-format请参阅自述文件中的此部分： [https：//github.com/NVlabs/ProtoMotions？tab](https://github.com/NVlabs/ProtoMotions?tab=readme-ov-file#convert-the-motions-to-motionlib-format) =readme-ov-file#convert-the-motions-to-motionlib-format
> > You need to save your file in the AMASS format and body ordering. Then, you run `convert_amass_to_isaac` (you might need to slightly tweak it to load your specific file) which will re-order the joints and convert everything to our format. This also handles converting the humanoid to z-axis up.您需要以 AMASS 格式和身体顺序保存文件。然后，运行 `convert_amass_to_isaac` （您可能需要稍微调整一下才能加载您的特定文件），它将重新排序关节并将所有内容转换为我们的格式。这还可以将人形转换为 z 轴向上。
> 
> Thank you for your advice , i do make some progress.谢谢你的建议，我确实取得了一些进步。 First i save my file in the AMASS format, and run covert_amass_to_isaac.py, then i run play_motion.py, the robot appears to be floating in the air and facing the ground, as shown in the following image:首先我将文件保存为 AMASS 格式，然后运行 ​​covert_amass_to_isaac.py，然后运行 ​​play_motion.py，机器人似乎漂浮在空中并面朝地面，如下图所示：
> 
> [image] I tried correction = R.from_euler('x', -90, degrees=True) ,and the result changed to this我尝试了 correction = R.from_euler('x', -90, degrees=True)，结果变成了这样
> 
> [image] In both of these two situations, my robots seems to remain in the air.在这两种情况下，我的机器人似乎都停留在空中。 Could you give me some advice? Or do you think the issue still lies in the conversion of my data into the AMASS format?您能给我一些建议吗？或者您认为问题仍然在于将我的数据转换为 AMASS 格式？

have you sloved this problem? I met this problem, too

sunmengmm, 2025-03-31:
> The default AMASS format is Y-up. If your data is in Z-up already, then you can comment out [...]

Thank you very much, I solved the problem

closed 2025-03-31 by sunmengmm

Gongkk, 2025-05-07:
> hi, thank u for your contribution! i do record human motion in a video, and I collect some data from TRAM(Global Trajectory and Motion of 3D Humans from in-the-wild Videos), its outputs include data like:hps_track.npy(pred_pose,pred_rotmat,pred_trans...) and camera.npy(world_cam_R,world_cam_T....)
> 
> AND i want to convert these outputs into a file like smpl_walk_humanoid.npy in /ProtoMotions/data/motions/.
> 
> Why the order of body_names in /ProtoMotions/protomotions/config/robot/smpl.yaml IS quite diffierent from the default order in the SMPL MODEL.
> 
> In smpl.yaml, you set body_names: ['Pelvis', 'L_Hip', 'L_Knee', 'L_Ankle', 'L_Toe', 'R_Hip', 'R_Knee', 'R_Ankle', 'R_Toe', 'Torso', 'Spine', 'Chest', 'Neck', 'Head', 'L_Thorax', 'L_Shoulder', 'L_Elbow', 'L_Wrist', 'L_Hand', 'R_Thorax', 'R_Shoulder', 'R_Elbow', 'R_Wrist', 'R_Hand'] BUT WE HAVE the default order in the SMPL MODEL:['Pelvis', 'L_Hip', 'R_Hip', 'Torso', 'L_Knee', 'R_Knee', 'Spine', 'L_Ankle', 'R_Ankle', 'Chest','L_Toe', 'R_Toe', 'Neck', 'L_Thorax', 'R_Thorax', 'Head', 'L_Shoulder', 'R_Shoulder', 'L_Elbow', 'R_Elbow', 'L_Wrist', 'R_Wrist', 'L_Hand', 'R_Hand'] Altough i did the permutations ,the play_motion.py shows my motion file may not be converted correctly.
> 
> Hope u can give me some advice in making motion files like smpl_walk_humanoid.npy. Thank you so much.

Hi, I am also working on this step of converting the TRAM output file to the AMASS data format. Can you pleas share some experience how you achieve this? It's greatly appreciated if you can show me your code. @sunmengmm