NVlabs/ProtoMotions #74: AMP Error

sherwin0402, 2025-03-26:
Thank you for your work. I am trying to run with just AMP and I use the following train code and eval code on my 4070. I believe that the training has converged judging by the discriminator l2_total but when I run the eval_agent, the model is not walking at all. I am not sure if I need to let it train longer, or there is something wrong with the eval_agent. A similar behaviour is also noticed when I train with humanoid Unitree H1.

Any help is appreciated, thanks!

```bash
PYTHON_PATH protomotions/train_agent.py +exp=amp_mlp motion_file=data/motions/smpl_humanoid_walk.npy +robot=smpl +simulator=isaacgym experiment_name=amp_smpl_walk +opt=wandb num_envs=512

PYTHON_PATH protomotions/eval_agent.py +robot=smpl +simulator=isaacgym +motion_file=data/motions/smpl_humanoid_walk.npy +checkpoint=results/amp_smpl_walk/last.ckpt
```

[image]

[image]

tesslerc, 2025-03-26:
I suggest training for longer.
It's hard to identify when AMP has converged, the metrics aren't very obvious.
From what I've seen, the only real indication is visual inspection.

As opposed to motion tracking where you can measure tracking performance.

sherwin0402, 2025-03-26:
Thanks for the quick reply! Is there a way to train with the GUI enabled? Or perhaps automatically take some periodic snapshots of the training?

sherwin0402, 2025-03-27:
Yes, I used the built-in h1_walk motion file (where the robot starts from center, and just walks around randomly). Similarly, my epoch passed quickly and the model doesn't stand up even after multiple rounds. I also tested the SMPL model and motion and they both work on play as animation but not after training. I am now following the recommendation to train longer.

lilinfuxle1, 2025-03-27:
> Yes, I used the built-in h1_walk motion file (where the robot starts from center, and just walks around randomly). Similarly, my epoch passed quickly and the model doesn't stand up even after multiple rounds. I also tested the SMPL model and motion and they both work on play as animation but not after training. I am now following the recommendation to train longer and will report back soon

I have tried to use a 3060 graphics card for training for about 12 hours, but it was basically impossible to achieve standing. Instead, the training results within a relatively short period of time enabled the model to stand for a short while.

tesslerc, 2025-03-27:
Could you try adding the flag to the end
```
+env.config.enable_height_termination=true
```
I see this flag is missing from the amp config and the behavior you're explaining aligns.

sherwin0402, 2025-03-28:
Hi @tesslerc, that fixed the problem for me. The Unitree H1 now walks with the default walking file. Can I check for more dynamic motions such as jumping, squatting, should this height_termination flag be set to true?

tesslerc, 2025-03-28:
Thanks for confirming!
The reason it wasn't on by default is that some motions can't succeed with it turned on. Such as crawling and cartwheels where the hands have to touch the ground.

AMP in general lacks a good termination condition/warmup strategy to ensure the robot doesn't deviate too much from the data.

sherwin0402, 2025-03-28:
Thanks for the clarification and information! I will close this issue now.

closed 2025-03-28 by sherwin0402