NVlabs/ProtoMotions #77: Potential issue with local rotation in motion retarget pipeline

Daffan, 2025-04-02:
There might be a potential issue with computing local rotation in motion retargeting pipeline.

Robot's current local rotation is computed by always assuming y-axis joint (see mimic_utils.py)
https://github.com/NVlabs/ProtoMotions/blob/14368ee81ed2dd64c8de90808c3b98dfb641d01b/protomotions/envs/mimic/mimic_utils.py#L177

In retargeting pipeline, the local rotation is computed using actual joint axis (not always assuming y-axis)
https://github.com/NVlabs/ProtoMotions/blob/14368ee81ed2dd64c8de90808c3b98dfb641d01b/data/scripts/retargeting/mink_retarget.py#L301

Also, I am not sure why local rotation is not converted to proper kinematic? It also contains the dimensions for extended links and they are appended at the end.
https://github.com/NVlabs/ProtoMotions/blob/14368ee81ed2dd64c8de90808c3b98dfb641d01b/data/scripts/retargeting/torch_humanoid_batch.py#L278

I have been trying to retarget the motion to Booster T1 robot, but got very large `lr_err` during the training. This is what I think might cause the issue. Let me know your thought. Thank you!!

tesslerc, 2025-04-02:
Thanks! We would be very happy to accept a PR on this!

(1)
Yes, there seems to be a mismatch between how we map the axis in the retargeting script and the way it is done in the motion tracking logic. We haven't been using local rotation errors, which might explain why it has gone unnoticed.

Here all revolute joints are mapped to the y-axis. Which is a wrong assumption.
https://github.com/NVlabs/ProtoMotions/blob/efad4152ea8425cd0c5d5806e5352be4b3246908/protomotions/utils/motion_lib.py#L752

The correct approach is to extract the true axis from the skeleton like here
https://github.com/NVlabs/ProtoMotions/blob/efad4152ea8425cd0c5d5806e5352be4b3246908/data/scripts/retargeting/torch_humanoid_batch.py#L90
 and use that in both MotionLib (previous code block) and in the motion tracking code here:
https://github.com/NVlabs/ProtoMotions/blob/efad4152ea8425cd0c5d5806e5352be4b3246908/protomotions/envs/mimic/mimic_utils.py#L177

- simpler, but not as clean, is to just map all local rotations of revolute joints to the y-axis.

(2)
Regarding the local rotations and conversion to "proper kinematic". My suggestion here is to save the local rotations without the extra (fixed) joints. This will make the local rotations consistent with the dofs representation, which only takes into account actuated (not fixed) joints.

https://github.com/NVlabs/ProtoMotions/blob/efad4152ea8425cd0c5d5806e5352be4b3246908/protomotions/envs/mimic/env.py#L412

this computation converts dofs to local and the dimension for the local rots is `[envs, actuated joints, 3]`.

martinorrje, 2025-04-04:
I think there is a similar issue here: 

https://github.com/NVlabs/ProtoMotions/blob/efad4152ea8425cd0c5d5806e5352be4b3246908/protomotions/envs/base_env/env_utils/humanoid_utils.py#L59-L61

Where the y-axis is assumed to be the rotation axis for each joint.

tesslerc, 2025-04-07:
@martinorrje and @Daffan can you take a look [here](https://github.com/NVlabs/ProtoMotions/pull/81) and let me know what you think?
I'd like your approval before I merge.

Main changes are:
1. For consistency, we no longer use the local rotations from the motion preprocessing. Rather, we compute location rotations from the dof pos. This is a common method for both simulated and reference motions, ensuring consistency.
2. Added a field for joint_axis in the robot config. This defines for each joint which axis it acts upon. So a 3-dof joint will be "xyz" whereas a 1-joint dof will be "x" or "y" or "z", depending on which axis it operates.

Ideally, point 2 above should merge dofs that control the same body part and return the local rotation of the body part itself and not an individual tensor for each motor. But the current solution should at least provide consistent representation.

Based on my experiments with the G1, the local rotation errors seem reasonable.

tesslerc, 2025-04-09:
Fixed here: https://github.com/NVlabs/ProtoMotions/commit/0ae8e89421a7d38618ee9cbc3c59ced655b50a27

closed 2025-04-09 by tesslerc