NVlabs/ProtoMotions #80: something wrong when processing AMASS

Aron-64, 2025-04-07:
[image]
Nothing is output after running the python convert_amass_to_isaac.py “path/to/AMASS” command，which makes me confused

Aron-64, 2025-04-07:
[image]
well，i fixed the format problem，but still have problems.
wish your help

tesslerc, 2025-04-07:
Could you try with this file?
[LeftTurn01_stageii.npz.zip](https://github.com/user-attachments/files/19629778/LeftTurn01_stageii.npz.zip)

We use the AMASS-X files (SMPL-X format) to retarget to the H1/G1. I haven't tried with the SMPL format, so not sure what to expect there.

Aron-64, 2025-04-08:
Thanks for your reply. According to the file you gave me, I successfully output the corresponding data using the smplx format, which is good news. However, when relocating towards h1, it fails to output the corresponding data. I noticed that there is an error line that says no module named data, but there is __init__.py under data, which is unbelievable!
good news:
[image]

[image]
bad news:

[image]

[image]

tesslerc, 2025-04-08:
Can you try also adding `--humanoid-type=smplx` in the retargeting script?
Humanoid type tells it what is the format it is working with (smpl or smplx) and then the robot type tells it what target robot to convert to.

Aron-64, 2025-04-09:
That's my fault, I screenshot only part of it. Like you said, adding the **--humanoid-type=smplx** command I also tried, but still no output!

[image]

[image]

tesslerc, 2025-04-09:
It seems like it's processing the ACCAD folder, so the code is working. Only issue is folder mapping.

I think I understand the issue. Truly an oversight on our end.
It currently looks for this directory structure:
```
amass_data_path/
    -  dataset1_name/
        - internal1_name/
        - ...
    -  dataset2_name/
        - internal2_name/
        - ...
```
to make it work (with the current code), you need to create a subset directory inside `left` and it will catch the file.

I'll add this to our TODO list to improve. Also feel free to submit a PR.

Aron-64, 2025-04-09:
Yes, you're right. It only captures the appropriate data when I add a subdirectory. This has worked when outputting the smplx motion file. But when I try to relocate to the h1 robot, it fails. I noticed that when generating the motion file for the relocation, it reported an error, do you guys have a similar error when relocating or maybe I made a mistake when configuring the environment!

[image]

[image]

[image]

Aron-64, 2025-04-09:
here is the smpl-x motionlib file

[image]

Aron-64, 2025-04-10:
Awesome!!! We finally solved the problem, and after removing the try/except, I found the problem based on the reported error. 
The error **_TypeError: ‘type’ object is not subscriptable_** occurs on line 334 of the mink_retarget.py file, specifically due to the return type annotation **dict[str, bytes]** in the function **get_assets**. 
This is because in Python 3.8 and earlier, built-in types such as dict, list, and so on do not support generic type annotations (such as **dict[str, bytes]**) directly using square brackets [ ], so i try to import **Dict** from the **typing** module.

[image]

[image]

Aron-64, 2025-04-10:
emmm，now I'm trying to redirect this data to a robot (different from g1 or h1) and it's reporting an error showing that I have the wrong joint type, but I noticed that in the mjcf file you guys uploaded for h1, the joint type is used as cylinder, sphere, etc. May I ask how you guys solved this problem?

[image]

tesslerc, 2025-04-10:
This seems like an issue with the mjcf parser in dm control.
You can rename the joint type to hinge from revolute, that should fix the issue.

Aron-64, 2025-04-11:
emmm，it doesn't work.Maybe you can share the steps you guys take when retargeting h1. So that I know exactly what files I need and what code to change?

closed 2025-04-14 by Aron-64