NVlabs/ProtoMotions #82: Memory Limit issue

ghassen-fatnassi, 2025-04-11:
hey
Protomotions works fantastically with isaac gym , but since my hardware isn't that good , T1000 8GB VRAM , I keep getting the assert batch_size%minibatchsize ==0 since i try to reduce the num_envs to 256/128 to make it work on my machine
can u help me pinpoint the minimbatchsize parameter i should change , i already tinkered with some but seems like I'm doing it in the wrong config file
thanks a lot!

ghassen-fatnassi, 2025-04-12:
thanks a lot for your advice will try it , i also have another question if you don't mind , sometimes when isaacgym renders a complex env that requires a lot of computation+ graphics  my gpu can't keep up and i get a segmentation fault/cuda out of memory , is there a way to reduce the resolution of the isaacgym so i can decrease the load of graphics on my gpu ?

tesslerc, 2025-04-14:
Ive never tried, but I believe that should be possible.
If you know the IsaacGym commands, I can point you to where they should be called in our code.

ghassen-fatnassi, 2025-04-15:
hey thanks a lot for your help
I managed to get H100 access on DIgital ocean with github student pack and that is no longer an issue
I'm thinking of adapting the Nymeria dataset released by meta to protomotions to see how it does with somehow an excess of data
can u pinpoint to me to which part of the code i should read well so i can convert the nymeria multiple .npy files to the single .npy file of motions used in Protomotions any help would be much appreciated , i can provide some info about the Nymeria body motion metadata:

[image]

[image]

ghassen-fatnassi, 2025-04-16:
I'm in short having trouble to how to :
1/animate the motion files using poselib (just to see them in matplotlib)
2/convert nymeria bodymotion to a format like the current npy files in motions folder in protomotion 
3/animate the newly converted file again to verify correctness
any help would be much appreciated

tesslerc, 2025-04-17:
https://github.com/NVlabs/ProtoMotions/blob/main/data/scripts/convert_amass_to_isaac.py

@ghassen-fatnassi this is the main script you're looking for.
Here we take a motion in SMPL format and convert it to "isaac" = a format that we can load with PoseLib.

ghassen-fatnassi, 2025-04-17:
thanks a lot !

ghassen-fatnassi, 2025-04-18:
I think I have succeded in retargeting it , can i submit a pull request on this topic after I refactor some code?

tesslerc, 2025-04-18:
@ghassen-fatnassi that would be great! We encourage community contributions ♥️

ghassen-fatnassi, 2025-04-18:
okay thanks a lot!

closed 2025-04-18 by ghassen-fatnassi