NVlabs/ProtoMotions #84: Real robot application

ZhangLiang-Cencen, 2025-04-17:
Cool work！

Has anyone used this framework on a real robot?

I want to migrate this framework to a real robot, but many observations are not available on the actual robot. Is there any way to avoid this? 

Thank you!

tesslerc, 2025-04-17:
Our focus has been on simulation, so we haven't applied to a real robot yet. We are starting to look in the direction.

It should be fairly easy to adapt the observations to what is available on the real robot.
If you could provide a list of available observations, I would be happy to help with that.

ZhangLiang-Cencen, 2025-04-21:
Thanks for your reply！
We can provide the following observations:
**root angular velocity (in base frame)
root posture expressed by quaternion
feedback joint angle
feedback joint velocity（in joint space）
actions
reference joint angle**
Thanks for your help!

ZhangLiang-Cencen, 2025-04-21:
Some additional information is included in the emails

closed 2025-06-04 by tesslerc