NVlabs/ProtoMotions #85: text input

ghassen-fatnassi, 2025-04-18:
is there a method to do inference on text input? even if it's not out of the box , can i get some guidance on how to extend the current protomotions to support it?

tesslerc, 2025-04-21:
If you're referring specifically to MaskedMimic, then there is a way.
If you look at the [training yaml](https://raw.githubusercontent.com/NVlabs/ProtoMotions/refs/heads/main/data/yaml_files/amass_train.yaml) (for example), you'll see how we map 3 textual description to each motion.

Then in MotionLib you have a flag for [create_text_embeddings](https://github.com/NVlabs/ProtoMotions/blob/main/protomotions/utils/motion_lib.py#L110C13-L110C35) which when set to True will [preprocess the text](https://github.com/NVlabs/ProtoMotions/blob/94f02d4f4904110d6808fa1efb98480c53a13d59/protomotions/utils/motion_lib.py#L516) using XCLIP.

You can see how we manage the text embedding obs in the maskedmimic observation [component](https://github.com/NVlabs/ProtoMotions/blob/94f02d4f4904110d6808fa1efb98480c53a13d59/protomotions/envs/mimic/components/masked_mimic_obs.py#L111).

The final piece is to define the architecture such that it will process this new observation. For MaskedMimic, we set it up as an [additional encoder](https://github.com/NVlabs/ProtoMotions/blob/main/protomotions/config/agent/masked_mimic/models/text_conditioning.yaml) that connects to the transformer.

tesslerc, 2025-04-21:
Also, if you just want to run inference on MaskedMimic with text you can use this config:
https://github.com/NVlabs/ProtoMotions/blob/main/protomotions/config/opt/masked_mimic/constraints/no_constraint.yaml

`+opt=masked_mimic/constraints/no_constraint env.config.masked_mimic.masked_mimic_masking.motion_text_embeddings_visible_prob=1`

This will run maskedmimic without any joint constraints and will condition on text (if a text condition exists in the motionlib).

ghassen-fatnassi, 2025-04-22:
thanks a lot , sorry for the late response , closing this issue

closed 2025-04-22 by ghassen-fatnassi