NVlabs/ProtoMotions #90: success rate is always low for the second stage

Mael-zys, 2025-04-29:
Hi, thank you for open sourcing such a wonderful work.

I'm trying to train the second stage on the whole AMASS datasets, but the training curve is really weird, the success rate is always very low. Do you have any suggestions on this? 

[image]

I can reproduce the first stage, but the second stage got this weird curve. I also tried to train with your pretrained full-body tracking policy, still not working. This is my training command:
```bash
python protomotions/train_agent.py +exp=masked_mimic/flat_terrain +robot=smpl \
+simulator=isaacgym motion_file=amass_data_proto/amass_train.pt agent.config.expert_model_path=data/pretrained_models/motion_tracker/smpl \
+terrain=flat +experiment_name=masked_mimic_amass_wandb_pretrain ngpu=4 eval_overrides.ngpu=4 num_envs=2048 agent.config.batch_size=8192 +opt=wandb
```

Btw could you provide the training curve for reference?

tesslerc, 2025-04-30:
Indeed this does not seem right.
Do the other metrics look ok?
Episode length / reward components

This is an example training run
[image]

Mael-zys, 2025-04-30:
Thanks a lot.

The other metrics also seems not correct. 

Btw, I'm using the code version 3.3. Maybe it's better to use the latest version?

[image]

[image]

Mael-zys, 2025-05-02:
I reinstalled the environment and re-processed the dataset. Now it is working. Thanks a lot!

[image]

closed 2025-05-02 by Mael-zys

Mael-zys, 2025-05-03:
Sorry, the successful case above is using one GPU to debug, but when using multiple GPUs, it fails to load the expert. This is the reason why the reward and success rate is always very low.

I found the real problem is the loading part: https://github.com/NVlabs/ProtoMotions/blob/main/protomotions/agents/masked_mimic/agent.py#L111-L114. 
In my case, with one node-multi-gpu, I should first load the model and then setup fabric.  

Now I can get a reasonable curve with multi-gpus. I'll also create a pull request to fix this.

```bash
python protomotions/train_agent.py +exp=masked_mimic/flat_terrain +robot=smpl \
+simulator=isaacgym motion_file=amass_data_proto/amass_train.pt agent.config.expert_model_path=data/pretrained_models/motion_tracker/smpl \
+terrain=flat +experiment_name=masked_mimic_amass_wandb_pretrain ngpu=4 eval_overrides.ngpu=4 num_envs=2048 agent.config.batch_size=8192 +opt=wandb
```

[image]

tesslerc, 2025-05-03:
That's really interesting.
Can you share how you run the multi GPU experiment?

I've had no issues so far with multi GPU and multi node.
I run each GPU as an independent process. Do you spawn a single proces?

Mael-zys, 2025-05-03:
This is the full command that I used:

```bash
export LD_LIBRARY_PATH=${CONDA_PREFIX}/lib/
export CUDA_VISIBLE_DEVICES=1,2,3,4

python protomotions/train_agent.py +exp=masked_mimic/flat_terrain +robot=smpl \
+simulator=isaacgym motion_file=amass_data_proto/amass_train.pt agent.config.expert_model_path=data/pretrained_models/motion_tracker/smpl \
+terrain=flat +experiment_name=masked_mimic_amass_wandb_pretrain ngpu=4 eval_overrides.ngpu=4 num_envs=2048 agent.config.batch_size=8192 +opt=wandb
```

I can also get separate processes:
[image]

This is part of the running info:

Initializing distributed: GLOBAL_RANK: 0, MEMBER: 1/4                                                                             
Importing module 'gym_38' (/home/yangsong.zhang/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)                        
Importing module 'gym_38' (/home/yangsong.zhang/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)                        
Importing module 'gym_38' (/home/yangsong.zhang/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)                        
Setting GYM_USD_PLUG_INFO_PATH to /home/yangsong.zhang/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json          
Setting GYM_USD_PLUG_INFO_PATH to /home/yangsong.zhang/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json          
Setting GYM_USD_PLUG_INFO_PATH to /home/yangsong.zhang/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json          
2025-05-04 01:35:46,119 - INFO - logger - logger initialized                                                                      
INFO:poselib:logger initialized                                                                                                                                                                                                   
2025-05-04 01:35:46,508 - INFO - logger - logger initialized                                                                      
INFO:poselib:logger initialized                                                                                                   
Initializing distributed: GLOBAL_RANK: 3, MEMBER: 4/4
[rank: 3] Seed set to 3
Setting seed: 3
Initializing distributed: GLOBAL_RANK: 1, MEMBER: 2/4
[rank: 1] Seed set to 1
Setting seed: 1
2025-05-04 01:35:47,239 - INFO - logger - logger initialized
INFO:poselib:logger initialized
Optimizing mesh by merging flat regions...
Optimized mesh vertex count: 42
Note: 'sim' is not a field in RobotConfig, it will be ignored
Optimized mesh vertex count: 42
PyTorch version 2.2.0+cu121
Device count 4
/home/yangsong.zhang/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /data3/yangsong/.cache/torch_extensions/py38_cu121 as PyTorch extensions root...
Note: 'sim' is not a field in RobotConfig, it will be ignored
Initializing distributed: GLOBAL_RANK: 2, MEMBER: 3/4
[rank: 2] Seed set to 2
Setting seed: 2
[2025-05-04 01:35:48,342][lightning_fabric.utilities.rank_zero][INFO] - ----------------------------------------------------------
------------------------------------------
distributed_backend=nccl
All distributed processes registered. Starting with 4 processes
----------------------------------------------------------------------------------------------------

[rank: 0] Seed set to 0                                                                                                           
Setting seed: 0                                                                                                                   
Emitting ninja build file /data3/yangsong/.cache/torch_extensions/py38_cu121/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
Not connected to PVD
+++ Using GPU PhysX
Initial heightfield size: 2800x2400 = 6720000 vertices
Initial heightfield size: 2800x2400 = 6720000 vertices
PyTorch version 2.2.0+cu121
Device count 4
/home/yangsong.zhang/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /data3/yangsong/.cache/torch_extensions/py38_cu121 as PyTorch extensions root...
Emitting ninja build file /data3/yangsong/.cache/torch_extensions/py38_cu121/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
Optimizing mesh by merging flat regions...
ninja: no work to do.
Loading extension module gymtorch...
Not connected to PVD
+++ Using GPU PhysX
Optimizing mesh by merging flat regions...
Optimized mesh vertex count: 42
Note: 'sim' is not a field in RobotConfig, it will be ignored
Optimized mesh vertex count: 42
Note: 'sim' is not a field in RobotConfig, it will be ignored
Physics Engine: PhysX
Physics Device: cuda:1
GPU Pipeline: enabled
Adding terrain
Terrain added
Creating 1024 environments... ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:--PyTorch version 2.2.0+cu121
Device count 4
/home/yangsong.zhang/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /data3/yangsong/.cache/torch_extensions/py38_cu121 as PyTorch extensions root...
Emitting ninja build file /data3/yangsong/.cache/torch_extensions/py38_cu121/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)

Mael-zys, 2025-05-03:
Here is an example curve though using other data. The green and blue curves correspond to the experiments before and after modifying the loading orders: 

[image]

This is the code that I modified: https://github.com/NVlabs/ProtoMotions/pull/95/files