NVlabs/ProtoMotions #94: Recording Full Body Tracking Motions in Headless Setup

echen4628, 2025-05-01:
Hi, I am wondering how to record videos in a headless setting. I have trained a policy for full body tracking (part 1), and I want to visualize its performance. Right now, I am running 

```
PYTHON_PATH protomotions/eval_agent.py +robot=smpl +simulator=isaacgym +motion_file=data/motions/smpl_humanoid_walk.npy +checkpoint=results/mlp_full_body_tracker/last.ckpt +headless=True
```
which produces the following output and continues running
```
Importing module 'gym_38' (/home/ubuntu/ProtoMotions/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/ubuntu/ProtoMotions/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
2025-05-01 21:08:46,834 - INFO - logger - logger initialized
INFO:poselib:logger initialized
protomotions/eval_agent.py:65: UserWarning: 
The version_base parameter is not specified.
Please specify a compatability version level, or None.
Will assume defaults for version 1.1
  @hydra.main(config_path="config")
/home/ubuntu/ProtoMotions/env/lib/python3.8/site-packages/hydra/_internal/hydra.py:119: UserWarning: Future Hydra versions will no longer change working directory at job runtime by default.
See https://hydra.cc/docs/1.2/upgrades/1.1_to_1.2/changes_to_job_working_dir/ for more information.
  ret = run_job(
Loading training config file from results/mlp_full_body_tracker/config.yaml
[2025-05-01 21:08:47,518][lightning_fabric.utilities.rank_zero][INFO] - You are using a CUDA device ('NVIDIA A10G') that has Tensor Cores. To properly utilize them, you should set `torch.set_float32_matmul_precision('medium' | 'high')` which will trade-off precision for performance. For more details, read https://pytorch.org/docs/stable/generated/torch.set_float32_matmul_precision.html#torch.set_float32_matmul_precision
Initializing distributed: GLOBAL_RANK: 0, MEMBER: 1/1
[2025-05-01 21:08:47,520][lightning_fabric.utilities.rank_zero][INFO] - ----------------------------------------------------------------------------------------------------
distributed_backend=nccl
All distributed processes registered. Starting with 1 processes
----------------------------------------------------------------------------------------------------

/home/ubuntu/ProtoMotions/env/lib/python3.8/site-packages/torch/functional.py:507: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at ../aten/src/ATen/native/TensorShape.cpp:3549.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
Initial heightfield size: 2800x1150 = 3220000 vertices
Optimizing mesh by merging flat regions...
Optimized mesh vertex count: 28
Note: 'sim' is not a field in RobotConfig, it will be ignored
PyTorch version 2.2.0+cu121
Device count 1
/home/ubuntu/ProtoMotions/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/ubuntu/.cache/torch_extensions/py38_cu121 as PyTorch extensions root...
Emitting ninja build file /home/ubuntu/.cache/torch_extensions/py38_cu121/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: enabled
Adding terrain
Terrain added
Creating 1 environments... ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 100% 0:00:00
/home/ubuntu/ProtoMotions/protomotions/utils/motion_lib.py:127: UserWarning: To copy construct from a tensor, it is recommended to use sourceTensor.clone().detach() or sourceTensor.clone().detach().requires_grad_(True), rather than torch.tensor(sourceTensor).
  torch.tensor(key_body_ids, dtype=torch.long, device=device),
Loading motions from yaml/npy file
Loading 1/1 motion files: data/motions/smpl_humanoid_walk.npy
Loaded 1 motions with a total length of 5.667s.
Loading model from checkpoint: /home/ubuntu/ProtoMotions/results/mlp_full_body_tracker/lightning_logs/version_0/last.ckpt
Loading env checkpoint: /home/ubuntu/ProtoMotions/results/mlp_full_body_tracker/lightning_logs/version_0/env_0.ckpt
/home/ubuntu/ProtoMotions/protomotions/envs/mimic/env.py:492: UserWarning: std(): degrees of freedom is <= 0. Correction should be strictly less than the reduction factor (input numel divided by output numel). (Triggered internally at ../aten/src/ATen/native/ReduceOps.cpp:1760.)
  self.log_dict[f"raw/{rew_name}_std"] = rew.std()
/home/ubuntu/ProtoMotions/protomotions/envs/mimic/env.py:496: UserWarning: std(): degrees of freedom is <= 0. Correction should be strictly less than the reduction factor (input numel divided by output numel). (Triggered internally at ../aten/src/ATen/native/ReduceOps.cpp:1760.)
  self.log_dict[f"scaled/{rew_name}_std"] = rew.std()
/home/ubuntu/ProtoMotions/protomotions/envs/mimic/env.py:514: UserWarning: std(): degrees of freedom is <= 0. Correction should be strictly less than the reduction factor (input numel divided by output numel). (Triggered internally at ../aten/src/ATen/native/ReduceOps.cpp:1760.)
  self.log_dict[f"mimic_other/{rew_name}_std"] = rew.std()
```

It seems like the motion is being evaluated? It's just that I cannot see anything.

I noticed that the record shortcut "L" is only meant for the viewer, according to this [line](https://github.com/NVlabs/ProtoMotions/blob/main/protomotions/simulator/isaacgym/simulator.py#L94). 

Do you have any suggestions for recording inference motion in a headless setup? Thanks in advance!

tesslerc, 2025-05-02:
The key-based commands are for interactive control.
It's possible to add similar functionality such that it loads the simulator and records several episodes automatically.

If you have an example of headless recording for IsaacGym, we can add that functionality.

echen4628, 2025-05-04:
Hi, sorry I missed this reply. Thanks for the info! I don't have an example for headless recording yet. Just double-checking, if I want to record motions in IsaacGym, I would need to add that feature?

tesslerc, 2025-05-09:
There might be other ways. Such as streaming the humanoid's pose and then rendering it using external tools.
But to directly record using IsaacGym, I believe you'd need to spawn cameras (no viewport camera in headless mode), make sure they render even in headless mode, and then record their data.

closed 2025-05-09 by echen4628