NVlabs/ProtoMotions #97: Help using LAFAN1 dataset with G1 robot format

learning-github-zgy, 2025-05-07:
Hi, thank you very much for your great work!

I’m currently trying to train a policy based on the G1 robot model. I have access to the publicly available [LAFAN1](https://huggingface.co/datasets/lvhaidong/LAFAN1_Retargeting_Dataset) dataset, but the data is in CSV format and represents motions in the G1 robot’s joint configuration, not in SMPL format.

I noticed that your project uses AMASS-formatted mocap data. Since my data is in a different format, could you kindly provide some guidance or suggestions on how I could preprocess or adapt this dataset to be used with your framework?

Any pointers to relevant scripts, conversion pipelines, or interface expectations in your codebase would be greatly appreciated.

Thank you in advance!

tesslerc, 2025-05-13:
You can see [this](https://github.com/NVlabs/ProtoMotions/blob/main/data/scripts/retargeting/mink_retarget.py) code for how we retarget the AMASS data onto the G1/H1 morphology + Isaac format.

We haven't used the LAFAN1 dataset, so I can't comment on the specifics.

learning-github-zgy, 2025-05-15:
Okay, I'll try following this code logic. Thank you for your response.

closed 2025-05-15 by learning-github-zgy